Botspot/wor-flasher #3: USB boot support

pranavr2003, 2021-08-31:
Right now the README only mentions support for using this with an SD card. However, USB 3.0 sticks are faster (not as much as an SSD, of course), so are there any plans to support USB flash and boot from it?

pranavr2003, 2021-09-01:
@Botspot have a look at this:
[image]

This error omes right after selecting the version of Windows and my 64 gig SanDisk USB 3.0 drive.

Botspot, 2021-09-01:
Looks like you're having the same problem as [this person](https://github.com/Botspot/wor-flasher/issues/4#issuecomment-909701542).

Botspot, 2021-09-01:
What language is your Raspberry Pi OS using?

pranavr2003, 2021-09-01:
@Botspot Yeah, I checked that one out too. I guess USB boot is the issue.

pranavr2003, 2021-09-01:
@Botspot Usual, English (US)

Botspot, 2021-09-01:
> @Botspot Yeah, I checked that one out too. I guess USB boot is the issue.

It shouldn't be.

Botspot, 2021-09-01:
The dialog window failed to launch, that's what happened. It appears this is being caused by some sort of unsupported character.

pranavr2003, 2021-09-01:
@Botspot Well I don't know about that, my timezone is set to United States, and the lang is Eng, US

pranavr2003, 2021-09-01:
I did a fresh install of RPi-OS for this, so I haven't messed around with anything in the background

Botspot, 2021-09-01:
> @Botspot Well I don't know about that, my timezone is set to United States, and the lang is Eng, US

I want you to change your script to see if we can identify the issue.

Starting on line 186 of install-wor-gui.sh, replace this code:
```
CONFIG_TXT="$(yad "${yadflags[@]}" --width=530 --height=420 --image="$DIRECTORY/logo-full.png" \
  --text="$window_text" --separator='\n' --form \
  "${existing_img_chk[@]}" \
  --field='Edit <b>config.txt</b>:':TXT "$CONFIG_TXT" \
  --field="<b>Warning!</b> All data on the target drive will be deleted!":LBL '' \
  --button='<b>Flash</b>'!!"Warning! All data on the target drive will be deleted! Backup any files before it's too late!":0
)"
```
With this:
```
echo "yad ${yadflags[@]} --width=530 --height=420 --image=$DIRECTORY/logo-full.png \
  --text=$window_text --separator='\n' --form \
  ${existing_img_chk[@]} \
  --field='Edit <b>config.txt</b>:':TXT $CONFIG_TXT \
  --field=<b>Warning!</b> All data on the target drive will be deleted!:LBL '' \
  --button='<b>Flash</b>'!!Warning! All data on the target drive will be deleted! Backup any files before it's too late!:0"
exit 0
```
This replacement code will output the entire terminal command used to launch the final window.

pranavr2003, 2021-09-01:
Sure

pranavr2003, 2021-09-01:
@Botspot 
```
pi@raspberrypi:~ $ LANG=C LC_ALL=C LANGUAGE=C ~/wor-flasher/install-wor-gui.sh
DL_DIR: /home/pi/wor-flasher-files
DIRECTORY: /home/pi/wor-flasher
UUID: 6f7de912-4143-431b-b605-924c22ab9b1f
RPI_MODEL: 4
WIN_LANG: en-us
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 1
yad --center --width=310 --height=250 --window-icon=/logo.png --title=Windows on Raspberry --width=530 --height=420 --image=/logo-full.png   --text=<big><b>Installation Overview</b></big>

\ufffd\ufffd\ufffd Target drive: <b>/dev/sda</b> (57.3GB  USB  SanDisk 3.2Gen1)
\ufffd\ufffd\ufffd Drive is larger than 25 GB - can install Windows on itself
\ufffd\ufffd\ufffd Hardware type: <b>Raspberry Pi 4</b>
\ufffd\ufffd\ufffd Operating system: <b>Windows 11 (en-us) arm64 build 22000.168</b> --separator='\n' --form      --field='Edit <b>config.txt</b>:':TXT 

# don't change anything below this point #
arm_64bit=1
enable_uart=1
uart_2ndstage=1
enable_gic=1
armstub=RPI_EFI.fd
disable_commandline_tags=1
disable_overscan=1
device_tree_address=0x1f0000
device_tree_end=0x200000
dtoverlay=miniuart-bt   --field=<b>Warning!</b> All data on the target drive will be deleted!:LBL ''   --button='<b>Flash</b>'!!Warning! All data on the target drive will be deleted! Backup any files before it's too late!:0
```

pranavr2003, 2021-09-01:
@Botspot I tried installing Rpi-OS on the drive and then used wor-flasher, but it gives the same error. I then formatted it using Imager and tried again, but couldn't get through.

Botspot, 2021-09-01:
That looks fine. I just uploaded a fix for the DIRECTORY variable assignment code. WoR-flasher should self-update when you run it again, but your manual edits won't be overwritten so revert the file first.

I'm not sure if the fix I made fixes your particular error or not, but it certainly fixes [this one](https://github.com/Botspot/wor-flasher/issues/5).

Botspot, 2021-09-01:
> @Botspot I tried installing Rpi-OS on the drive and then used wor-flasher, but it gives the same error. I then formatted it using Imager and tried again, but couldn't get through. All my packages are updated too

I admire your effort, but that should never need to be necessary to make it work. Say the bug disappeared - now instead of it being traced and fixed, it would disappear, waiting for the next person!

pranavr2003, 2021-09-01:
@Botspot The issue is that it looks fine, when it probably shouldn't, so I'm not able to troubleshoot the code myself

pranavr2003, 2021-09-01:
@Botspot Exactly, I thought so too

pranavr2003, 2021-09-01:
@Botspot Surely this couldn't be due to the fact that I'm using a Pi 400? The first GUI window says Pi 4/400.

Botspot, 2021-09-01:
> @Botspot Surely this couldn't be due to the fact that I'm using a Pi 400? The first GUI window says Pi 4/400.

No, this is clearly a bug in my code. Not your USB setup, not your hardware.

pranavr2003, 2021-09-01:
@Botspot 
```
pi@raspberrypi:~ $ LANG=C LC_ALL=C LANGUAGE=C ~/wor-flasher/install-wor-gui.sh
DL_DIR: /home/pi/wor-flasher-files
DIRECTORY: /home/pi/wor-flasher
UUID: 6f7de912-4143-431b-b605-924c22ab9b1f
RPI_MODEL: 4
WIN_LANG: en-us
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 1
Unable to parse command line: Invalid byte sequence in conversion input
User exited when reviewing information and customizing config.txt
```
The same error

Botspot, 2021-09-01:
I still haven't been able to reproduce this on my system. (It's been customized over the years, maybe that's why?)
In any case, I've tried replacing the Unicode bullet point characters (•) with dashes. Let's see if that makes your dialog window launch successfully.

pranavr2003, 2021-09-01:
I'll try again

pranavr2003, 2021-09-01:
It does go through this time around!

pranavr2003, 2021-09-01:
So I guess there was with the UFT-8 encoding?

Botspot, 2021-09-01:
> It does go through this time around!

Excellent!!! Looks like something, somewhere on your OS does not work correctly when a non-ASCII character is encountered.

Botspot, 2021-09-01:
In any case, thank you so much for your patience and persistence! Your effort has benefited many others.

closed 2021-09-01 by Botspot