Botspot/wor-flasher #4: can't download the os

KolpixTV, 2021-08-31:
i can't download any os here is the log:

Input configuration:
DL_DIR: /home/pi/wor-flasher-files
RUN_MODE: gui
RPI_MODEL: 4
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 0
UUID: 6f7de912-4143-431b-b605-924c22ab9b1f
WIN_LANG: de-de
CONFIG_TXT: \u2935
  > # don't change anything below this point #
  > arm_64bit=1
  > enable_uart=1
  > uart_2ndstage=1
  > enable_gic=1
  > armstub=RPI_EFI.fd
  > disable_commandline_tags=1
  > disable_overscan=1
  > device_tree_address=0x1f0000
  > device_tree_end=0x200000
  > dtoverlay=miniuart-bt
CONFIG_TXT: \u2934

Downloading WoR PE-based installer from Google Drive
download_from_gdrive(): requires a Google Drive file UUID!
File UUID is the end of a sharable link: https://drive.google.com/uc?export=download&id=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

(zenity:17835): Gtk-WARNING **: 20:24:23.143: Failed to set text 'An error has occurred:
download_from_gdrive(): requires a Google Drive file UUID!
File UUID is the end of a sharable link: https://drive.google.com/uc?export=download&id=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Exiting now.' from markup due to error parsing markup: Fehler in Zeile 3: Entit\u00e4t endete nicht mit einem Semikolon; wahrscheinlich haben Sie ein &-Zeichen benutzt, ohne eine Entit\u00e4t begin

Botspot, 2021-08-31:
This appears to be an issue caused by your Internet connection.

KolpixTV, 2021-08-31:
i have internet on my raspberry pi i can watch youtube videos etc.

KolpixTV, 2021-08-31:
[image]

KolpixTV, 2021-08-31:
[image]

KolpixTV, 2021-08-31:
i just had to restart my raspberry pi now it works fine but thank you for your help :D

KolpixTV, 2021-08-31:
i thinked it worked because it was loading like 5 minutes but i have the same issue again

Botspot, 2021-08-31:
> i thinked it worked because it was loading like 5 minutes but i have the same issue again

The problem seems to reside in the way the script determines the Google Drive link. It's like the HTTP redirection is not working, or it's not being reported properly.
@KolpixTV, could you run this command in a terminal and report the output?
```
wget --spider --content-disposition --trust-server-names -O /dev/null https://worproject.ml/dldserv/worpe/downloadlatest.php
```
A variant of that command can be found on line 344 of the `install-wor.sh` script. (note that line-numbers are subject to change if I add or subtract lines of code above that point)

KolpixTV, 2021-08-31:
this is the output of the command:
```

pi@raspberrypi:~ $ wget --spider --content-disposition --trust-server-names -O /dev/null https://worproject.ml/dldserv/worpe/downloadlatest.php
Spider-Modus eingeschaltet. Es wird gepr\u00fcft, ob die Datei auf dem Server existiert.
--2021-08-31 22:13:23--  https://worproject.ml/dldserv/worpe/downloadlatest.php
Aufl\u00f6sen des Hostnamens worproject.ml (worproject.ml)\u2026 2606:4700:3037::ac43:a54a, 2606:4700:3037::6815:41b4, 172.67.165.74, ...
Verbindungsaufbau zu worproject.ml (worproject.ml)|2606:4700:3037::ac43:a54a|:443 \u2026 verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet \u2026 302 Found
Platz: https://dl.orangedox.com/WoR-PE-Package-1.0.0?dl=1 [folgend]
Spider-Modus eingeschaltet. Es wird gepr\u00fcft, ob die Datei auf dem Server existiert.
--2021-08-31 22:13:23--  https://dl.orangedox.com/WoR-PE-Package-1.0.0?dl=1
Aufl\u00f6sen des Hostnamens dl.orangedox.com (dl.orangedox.com)\u2026 35.81.40.28, 52.41.249.230, 34.208.169.147
Verbindungsaufbau zu dl.orangedox.com (dl.orangedox.com)|35.81.40.28|:443 \u2026 verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet \u2026 302 Found
Platz: https://drive.google.com/uc?export=download&id=1WHyHFYjM4WPAAGH2PICGEhT4R5TlxlJC [folgend]
Spider-Modus eingeschaltet. Es wird gepr\u00fcft, ob die Datei auf dem Server existiert.
--2021-08-31 22:13:24--  https://drive.google.com/uc?export=download&id=1WHyHFYjM4WPAAGH2PICGEhT4R5TlxlJC
Aufl\u00f6sen des Hostnamens drive.google.com (drive.google.com)\u2026 2a00:1450:4001:80f::200e, 142.250.185.110
Verbindungsaufbau zu drive.google.com (drive.google.com)|2a00:1450:4001:80f::200e|:443 \u2026 verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet \u2026 200 OK
L\u00e4nge: nicht spezifiziert [text/html]
Datei auf dem Server existiert und k\u00f6nnte weitere Verweise enthalten,
aber Rekursion ist abgeschaltet -- kein Download.
```

sorry but this is german i can set it to english if you want

Botspot, 2021-08-31:
> sorry but this is german i can set it to english if you want

No, actually, I think this is the reason why it did not work! My script is looking for this word in the output: "`Following`". In a German output, this word will never appear. Mystery solved!

Try running this and see if the output is in English this time:
```
LANG=C
LC_ALL=C
LANGUAGE=C
wget --spider --content-disposition --trust-server-names -O /dev/null https://worproject.ml/dldserv/worpe/downloadlatest.php
```

KolpixTV, 2021-08-31:
```
pi@raspberrypi:~/wor-flasher $ wget --spider --content-disposition --trust-server-names -O /dev/null https://worproject.ml/dldserv/worpe/downloadlatest.php
Spider mode enabled. Check if remote file exists.
--2021-08-31 22:23:10--  https://worproject.ml/dldserv/worpe/downloadlatest.php
Resolving worproject.ml (worproject.ml)... 2606:4700:3037::ac43:a54a, 2606:4700:3037::6815:41b4, 104.21.65.180, ...
Connecting to worproject.ml (worproject.ml)|2606:4700:3037::ac43:a54a|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://dl.orangedox.com/WoR-PE-Package-1.0.0?dl=1 [following]
Spider mode enabled. Check if remote file exists.
--2021-08-31 22:23:10--  https://dl.orangedox.com/WoR-PE-Package-1.0.0?dl=1
Resolving dl.orangedox.com (dl.orangedox.com)... 35.81.40.28, 34.208.169.147, 52.41.249.230
Connecting to dl.orangedox.com (dl.orangedox.com)|35.81.40.28|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://drive.google.com/uc?export=download&id=1WHyHFYjM4WPAAGH2PICGEhT4R5TlxlJC [following]
Spider mode enabled. Check if remote file exists.
--2021-08-31 22:23:11--  https://drive.google.com/uc?export=download&id=1WHyHFYjM4WPAAGH2PICGEhT4R5TlxlJC
Resolving drive.google.com (drive.google.com)... 2a00:1450:4001:80f::200e, 142.250.185.110
Connecting to drive.google.com (drive.google.com)|2a00:1450:4001:80f::200e|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.

```

This is the output now and now i get this if i want to isntall an os.

```
pi@raspberrypi:~/wor-flasher $ ~/wor-flasher/install-wor-gui.sh
DL_DIR: /home/pi/wor-flasher-files
UUID: 6f7de912-4143-431b-b605-924c22ab9b1f
RPI_MODEL: 4
WIN_LANG: de-de
cat: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/product': No such file or directory
cat: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/manufacturer': No such file or directory
DEVICE: /dev/sda
Drive /dev/sda is too small to install windows on itself. Using recovery-disk mode to install Windows on other larger devices.
CAN_INSTALL_ON_SAME_DRIVE: 0
cat: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/product': No such file or directory
cat: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/manufacturer': No such file or directory
Unable to parse command line: Invalid byte sequence in conversion input
User exited when choosing custom windows build ID

```

Botspot, 2021-08-31:
> now i get this output

Looks great!

> This is the output now and now i get this if i want to install an os.
> 
> ```
> pi@raspberrypi:~/wor-flasher $ ~/wor-flasher/install-wor-gui.sh
> DL_DIR: /home/pi/wor-flasher-files
> UUID: 6f7de912-4143-431b-b605-924c22ab9b1f
> RPI_MODEL: 4
> WIN_LANG: de-de
> cat: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/product': No such file or directory
> cat: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/manufacturer': No such file or directory
> DEVICE: /dev/sda
> Drive /dev/sda is too small to install windows on itself. Using recovery-disk mode to install Windows on other larger devices.
> CAN_INSTALL_ON_SAME_DRIVE: 0
> cat: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/product': No such file or directory
> cat: '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/manufacturer': No such file or directory
> Unable to parse command line: Invalid byte sequence in conversion input
> User exited when choosing custom windows build ID
> ```

This is an entirely different issue. Looks like the USB device you are using **does not have a name**. During testing, we never came across a storage drive that was not named like this.
I'm not sure if this is actually what's causing your problem though. Try this:

Find the `get-name` function in the `install-wor.sh` script. It looks like this:
```
get_name() { #get human-readable name of device: manufacturer and model name
  #Botspot made this by reverse-engineering the usb-devices command and udevadm commands.
  #input: /dev device
  [ -z "$1" ] && error "get_name(): requires an argument"
  [ ! -b "$1" ] && error "get_name(): Specified block device '$1' does not exist!"
  
  sys_path="$(find /sys/devices/platform -type d -name "$(basename "$1")")"
  #sys_path may be: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/host0/target0:0:0/0:0:0:0/block/sda
  
  if [ -z "$sys_path" ];then
    echo "get_name(): Failed to find a /sys/devices/platform entry for '$1'. Continuing." 1>&2
    return 1
  fi
  
  #Go up 6 directories:
  sys_path="$(echo "$sys_path" | tr '/' '\n' | head -n -6 | tr '\n' '/')"
  #sys_path may be: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/
  
  product="$(cat "${sys_path}product")"
  manufacturer="$(cat "${sys_path}manufacturer")"
  #serial="$(cat "$sys_path"/serial)"
  
  if [ "$manufacturer" != "$product" ];then
    echo "$manufacturer $product"
  else
    echo "$manufacturer"
  fi
}
```
Change that code to:
```
get_name() {
  echo "Storage Drive"
}
```
Save the file and try running it again.

KolpixTV, 2021-08-31:
```
pi@raspberrypi:~/wor-flasher $ ./install-wor-gui.sh 
DL_DIR: /home/pi/wor-flasher-files
UUID: 6f7de912-4143-431b-b605-924c22ab9b1f
RPI_MODEL: 4
WIN_LANG: de-de
DEVICE: /dev/sda
Drive /dev/sda is too small to install windows on itself. Using recovery-disk mode to install Windows on other larger devices.
CAN_INSTALL_ON_SAME_DRIVE: 0
Unable to parse command line: Invalid byte sequence in conversion input
**User exited when choosing custom windows build ID**

```
now i get this

Botspot, 2021-08-31:
> ```
> pi@raspberrypi:~/wor-flasher $ ./install-wor-gui.sh
> DL_DIR: /home/pi/wor-flasher-files
> UUID: 6f7de912-4143-431b-b605-924c22ab9b1f
> RPI_MODEL: 4
> WIN_LANG: de-de
> DEVICE: /dev/sda
> Drive /dev/sda is [...]

Looks like a nonexistant USB drive name had nothing to do with it, as the same error is there.
And I would suggest you run the script with full paths, but you already did that in [this earlier comment](https://github.com/Botspot/wor-flasher/issues/4#issuecomment-909606890).
This appears to be occurring when trying to specify a custom Windows build ID. Does that window appear, allowing you to paste in a custom build ID, or does it fail before then?

KolpixTV, 2021-08-31:
i can use a custom build ID but this also doesn't work:

```
pi@raspberrypi:~/wor-flasher $ ./install-wor-gui.sh 
DL_DIR: /home/pi/wor-flasher-files
UUID: a1b78728-530f-443f-b672-d742e7c14bcb
RPI_MODEL: 4
WIN_LANG: de-de
DEVICE: /dev/sda
Drive /dev/sda is too small to install windows on itself. Using recovery-disk mode to install Windows on other larger devices.
CAN_INSTALL_ON_SAME_DRIVE: 0
Unable to parse command line: Invalid byte sequence in conversion input
User exited when choosing custom windows build ID
pi@raspberrypi:~/wor-flasher $ 

```

Botspot, 2021-08-31:
> i can use a custom build ID but this also doesn't work:

Were you given the option to choose a custom build ID? Did you see a window that looked like this?
[image]

KolpixTV, 2021-08-31:
yes i can see this windows if i choose "custom""

Botspot, 2021-08-31:
> yes i can see this windows if i choose "custom""

Okay, so you can see that window. What text are you typing into that text box?

KolpixTV, 2021-08-31:
i choosed windows 11 but i type in "9be73aed-bec5-4fba-9b84-b917aa7cb366" if i use this

KolpixTV, 2021-08-31:
https://user-images.githubusercontent.com/68741965/131578529-ccd698de-21d6-4cfe-a518-b9d7b2e43761.mp4

https://user-images.githubusercontent.com/68741965/131578632-f4e1fbc3-f046-46e6-8381-7c997e182014.mp4

Botspot, 2021-08-31:
> i choosed windows 11 but i type in "9be73aed-bec5-4fba-9b84-b917aa7cb366" if i use this

When I first saw the "User exited when choosing custom windows build ID" error, I was looking through the code for creating the build-ID selection window. But it turns out I also placed this exact same error message at the ending review window! (I forgot to change the error message)

I've changed the script, but now that I know where to look the problem should be easier to track down. Also, thank you for your screen recordings.

Botspot, 2021-08-31:
After having some time to think about it, my best guess is that your locale is somehow inserting an unsupported unicode character into the final dialog box.
@KolpixTV, could you set the language to the C locale and then run the script?
```
LANG=C LC_ALL=C LANGUAGE=C ~/wor-flasher/install-wor-gui.sh
```

Botspot, 2021-09-01:
Update: [somebody else had the same issue.](https://github.com/Botspot/wor-flasher/issues/3) I narrowed it down to the Unicode bullet-point characters being used in the window. By replacing them with dash characters, the problem was fixed. Your problem ought to be fixed too, but before I close the issue I'd like your confirmation, @KolpixTV.

pranavr2003, 2021-09-01:
@Botspot This issue still seems to persist

[image]

Botspot, 2021-09-01:
> @Botspot This issue still seems to persist

It's great to see you've gotten that far! As the terminal opened this time, it means the previous error you encountered has now been fixed.
The new error you see is completely different.: looks like either a webserver issue or an Internet issue.

pranavr2003, 2021-09-01:
@Botspot Yeah, it must be a webserver issue. However that said, downloading from UUPDUMP is working fine on my Windows machine

Botspot, 2021-09-01:
> @Botspot Yeah, it must be a webserver issue. However that said, downloading from UUPDUMP is working fine on my Windows machine

Usually internet issues like this are not consistent. Do repeated attempts end in the same error every time?

pranavr2003, 2021-09-01:
@Botspot I guess the issue is now 'gone', but I'll keep you updated

Botspot, 2021-09-01:
> @Botspot I guess the issue is now 'gone', but I'll keep you updated

Good to know. And yes, while frustrating as Internet issues are, there's little to nothing the app developers can do. The UUP download portion of the script is already using the most reliable downloader available - `aria2c`.

pranavr2003, 2021-09-01:
@Botspot 

[image]

Bummer.

Botspot, 2021-09-01:
> @Botspot
> 
> [image]
> 
> Bummer.

Looks like your `udftools` package is missing?
On RPiOS, this package is installed by default.
Fix the issue with: `sudo apt install udftools` I'll put the same in the script as well.
Don't worry, subsequent re-runs will be much, much faster as your Windows image has already been generated.

Botspot, 2021-09-01:
IF that doesn't work, also try: `sudo modprobe udf`

pranavr2003, 2021-09-01:
Sure, I'll try that.
>Don't worry, subsequent re-runs will be much, much faster as your Windows image has already been generated.
Haha I wish, once I saw the mount error, I moved the ISO onto my Desktop, formatted the drive using Imager so I could put the ISO on WOR using my Windows machine. I tried a variety of things using `sudo mv` and `sudo cp`, only to realize that Imager formats the drive into a FAT32 partition, and `sudo mv/cp`'s limit is 4 GB, while the ISO is 4.2. By this time, I had totally messed up the partitions (less said about this, the better), and it took me a while to get it back to where it was. So I'm starting fresh again. Fingers crossed this time around after `udftools`

KolpixTV, 2021-09-01:
> Update: [somebody else had the same issue.](https://github.com/Botspot/wor-flasher/issues/3) I narrowed it down to the Unicode bullet-point characters being used in the window. By replacing them [...]

Hey,

sorry i was sleeping at this time i Check it out in Like 30 minutes

KolpixTV, 2021-09-01:
its working fine now Thank you for your help :D

closed 2021-09-01 by KolpixTV

pranavr2003, 2021-09-01:
@Botspot Works like a charm now. Turns out `udftools` was the culprit. Thanks.