Botspot/wor-flasher #5: Error: File doesn't exist

Gingerheadboy, 2021-08-31:
I click flash and this shows up on the terminal:
```
RPI_MODEL: 4
WIN_LANG: en-gb
DEVICE: /dev/sdb
CAN_INSTALL_ON_SAME_DRIVE: 1
CONFIG_TXT: \u2935
  > 
  > 
  > # don't change anything below this point #
  > arm_64bit=1
  > enable_uart=1
  > uart_2ndstage=1
  > enable_gic=1
  > armstub=RPI_EFI.fd
  > disable_commandline_tags=1
  > disable_overscan=1
  > device_tree_address=0x1f0000
  > device_tree_end=0x200000
  > dtoverlay=miniuart-bt
CONFIG_TXT: \u2934
Launching install-wor.sh in a separate terminal
/home/pi/wor-flasher/install-wor-gui.sh: line 231: /terminal-run: No such file or directory
The terminal running install-wor.sh has been closed.
            
```
do you have an idea what this means?
the file that it says doesn't exist, realy does exist.
@Botspot thank you so much for pi-apps and this tool!

Botspot, 2021-08-31:
> @Botspot thank you so much for pi-apps and this tool!

Your gratitude has been noted. ;)

> 
> ```
> Launching install-wor.sh in a separate terminal
> /home/pi/wor-flasher/install-wor-gui.sh: line 231: /terminal-run: No such file or directory
> The terminal running install-wor.sh has been closed.
> ```
> 
> do you have an idea what this means?
> the file that it says doesn't exist, realy does exist.

It appears that you ran the script incorrectly. This script must be run **with full paths.**

## Correct:
```
~/wor-flasher/install-wor-gui.sh
```
## Incorrect:
```
./wor-flasher/install-wor-gui.sh
```
and
```
cd ~/wor-flasher
./install-wor-gui.sh
```
and
```
cd ~/wor-flasher
install-wor-gui.sh
```

Gingerheadboy, 2021-08-31:
What you said is correct is exactly what I ran.

Gingerheadboy, 2021-08-31:
When I got the error I ran:

`~/wor-flasher/install-wor-gui.sh`

Botspot, 2021-08-31:
> When I got the error I ran:
> 
> `~/wor-flasher/install-wor-gui.sh`

Hmm. Somehow the DIRECTORY variable must have not been set. I have just made an update to the script. Now when you run it, it should display something like this:
```
DIRECTORY: /home/pi/wor-flasher
```

Gingerheadboy, 2021-08-31:
So if I rerun it, it will work?

Botspot, 2021-08-31:
> So if I rerun it, it will work?

I didn't attempt to fix anything because we don't know where the problem is yet.
If you re-run it, you may now see what the problem is. No matter what the outcome, the change I just made will help us to narrow down the problem.

Gingerheadboy, 2021-08-31:
Reinstalled, it nothing changes.

Botspot, 2021-08-31:
> Reinstalled, it nothing changes.

No need to actually click through the dialog boxes - the terminal output we want will be at the very beginning:
```
pi@botspot:~ $ /home/pi/wor-flasher/install-wor-gui.sh
DL_DIR: /home/pi/wor-flasher-files
DIRECTORY: /home/pi/wor-flasher
^C
```

Botspot, 2021-09-01:
I'm fairly sure that recent changes have fixed this issue. Could you confirm that, @Gingerheadboy?

Gingerheadboy, 2021-09-01:
I get the same error

Gingerheadboy, 2021-09-01:
But I did get the output the I was supposed to get.

Gingerheadboy, 2021-09-01:
Sorry, just reinstalled it now it does work😀. Thank you @Botspot for your hard work.
Can't wait to use Windows 11 for my first time.

Botspot, 2021-09-01:
> Confirmed, just reinstalled it now it does work😀. Thank you @Botspot for your hard work.
> Can't wait to use Windows 11 for my first time.

Excellent! Thank you for your patience.

closed 2021-09-01 by Botspot