Botspot/wor-flasher #6: WoR and additional components

bobster316, 2021-09-02:
First oiff i'd like to say excellent work 

When using rpi OS 32bit I had some additional componets running i.e.
i2c 0.96 OLED Display
LEDShim

Is there anyway i can get these to work after installing WoR.
Also has there been any further development to get wifi workling

Botspot, 2021-09-02:
> When using rpi OS 32bit I had some additional componets running i.e.
> i2c 0.96 OLED Display
> LEDShim
> 
> Is there anyway i can get these to work after installing WoR.

I would doubt there are drivers for these, but [here is the list](https://github.com/worproject/RPi-Windows-Drivers#status) of the current status of various WoR drivers.

> Also has there been any further development to get wifi workling

You can review the driver status as progress is continually made on them. Additionally, leepspvideo reportedly made a video that mentions some kind og WiFi solution. Here's the video: https://www.youtube.com/watch?v=lE1hI0Vql-U I haven't watched it.

bobster316, 2021-09-03:
Thanks will check it out

mobilegmYT, 2021-09-04:
Currently, there are no drivers for stuff like that on WoR, maybe you can talk to the manufacturer about making a ARM64 windows build of their drivers.

closed 2021-09-06 by Botspot