Botspot/wor-flasher #7: wor-flasher error during installation

bobster316, 2021-09-02:
I am getting the following error when running the installation 

An error has occurred 
Failed to generation a Windows ISO!
(uup_download_linux.sh script indicated a failure)
Exiting now

Botspot, 2021-09-02:
Either your Internet connection is unreliable, or you ran out of available disk space.

I will soon upload these changes:

- The uupdump script will retry 4 times before giving up.
- If you have insufficient disk space, an error will occur.

closed 2021-09-02 by Botspot

lobotech13, 2021-11-07:
I have the same problem with uup_download_linux.sh...
Impossible to generate the ISO.
Is it possible to generate the ISO manually ?

Botspot, 2021-11-09:
> I have the same problem with uup_download_linux.sh... Impossible to generate the ISO. Is it possible to generate the ISO manually ?

Sometimes the uupdump website is unstable. Usually you can solve this by specifying an older Windows Update ID.