Botspot/wor-flasher #8: Failed to generate ISO

petragom, 2021-09-02:
[image]

I’m trying to install but I get this error.

I’ve used a clean install of Raspberry OS 32 bit on a 16gb card and selected an external SSD of 256Gb to install it on.

(Thank you so much for making this!!! I’ve wanted to try WOR for so long, but couldn’t because the previous way requires a Windows installation)

Itai-Nelken, 2021-09-02:
You ran out if space. The windows iso is over 4gb so you need at least 5gb if free space.

pranavr2003, 2021-09-02:
@petragom It's either that you ran out of space, or it was a web server issue. For me, it was a web server issue, so it worked like a charm when I tried again

Botspot, 2021-09-02:
> @petragom It's either that you ran out of space, or it was a web server issue. For me, it was a web server issue, so it worked like a charm when I tried again

The error in this person's screenshot says "No space left on device", so it's definitely a storage space issue.

pranavr2003, 2021-09-02:
@Botspot Gosh I didn't see that, gotta watch it

Botspot, 2021-09-02:
If an image is being generated AND the free disk space is less than 7GB (the total size of uupdump folder after a flash), an error will occur saying:
```
Your system does not have enough usable disk space to generate a Windows image.
Please free up space or set the DL_DIR variable to a drive with more capacity.
7GB is necessary.
```

closed 2021-09-02 by Botspot