Botspot/wor-flasher #9: No flash button

nilmarsh555, 2021-09-02:
on installation oerview page, there is no flash button.

Botspot, 2021-09-02:
This final window is fairly tall. Are you sure your screen is tall enough to see the lowest edge of the window?

pranavr2003, 2021-09-02:
@nilmarsh555 Change your screen resolution temporarily. I had the same issue too.

Botspot, 2021-09-02:
> on installation oerview page, there is no flash button.

Helpful tip: Press and hold the **Alt** key and click and drag the window to move it. On smaller resolutions, I used this trick all the time.

Botspot, 2021-09-02:
This window was not originally that tall. But when did it change? Looks like it became really tall when I [replaced](https://github.com/Botspot/wor-flasher/commit/c74dd5608feab93a039211368346dfa7594a7e1f) the Unicode bullet points with dashes to fix issue #3. Weird. Looks like a bug, either in `yad` or GTK.

In any case, I've found a workaround that fixes this problem: By extending the top image. This prevents the window from being able to resize smaller horizontally, allowing its height to be reduced to a reasonable value.
Now the Installation Overview window looks like this:
[image]
I'll upload the changes soon.

Botspot, 2021-09-06:
These changes were uploaded a few days ago, so I'm closing the issue as the final window is not nearly as tall anymore, thereby allowing people on smaller resolutions to see and click the Flash button.

closed 2021-09-06 by Botspot