Botspot/wor-flasher #10: wor-flasher fails due to

bobster316, 2021-09-03:
/usr/bin/genisoimage: no space left on device. cannot fwrite 32768'1
failed to create ISO image

60.65% done, estimate finish Fri Sep  3 16:27:15 2021
/usr/bin/genisoimage: No space left on device. cannot fwrite 32768*1
Failed to create ISO image

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 1 of 4)
Retrieving aria2 script...

09/03 16:24:30 [NOTICE] Downloading 1 item(s)

bobster316, 2021-09-03:
Forgot to mention, i am using a 16GB Micro SD with Rasperry Pi OS 32Bit and a 
Sandisk 128GB USB Stick

bobster316, 2021-09-03:
Failed  attempt 2 of 4

Itai-Nelken, 2021-09-03:
you ran out of space, you need at least 7gb of free space to download windows.

bobster316, 2021-09-03:
My understanding is 16gb micro sd for raspberry pi os and 128gb usb stick for Windows 11 would be more than sufficient?

Itai-Nelken, 2021-09-03:
you need at least 7gb free on the 16gb sd card with rpiOS.

Itai-Nelken, 2021-09-03:
you can check disk usage in the file manager bottom bar I think.

bobster316, 2021-09-03:
Ok thanks, i will use a 32gb micro sd and try again
Thanks

Botspot, 2021-09-03:
> Ok thanks, i will use a 32gb micro sd and try again Thanks

The latest version of WoR-flasher is supposed to check for 7GB of free space before proceeding. Somehow, it must have not triggered for you.

roblatour, 2021-09-04:
After a couple of hours I got the same error, using a 16GB card.  Will try again with a 32GB card.  Downloaded the wor-flasher earlier today.

Botspot, 2021-09-04:
> After a couple of hours I got the same error, using a 16GB card. Will try again with a 32GB card. Downloaded the wor-flasher earlier today.

Strange. I'll bump up the space threshold from 7GB to 9GB.

roblatour, 2021-09-05:
worked fine with a 32gb card.  Thanks for all the work that has gone into this.

Botspot, 2021-09-06:
Excellent! I'm pleased to know this issue has been resolved. Enjoy running Windows!

closed 2021-09-06 by Botspot