Botspot/wor-flasher #13: "pnp driver watchdog" error on boot

TheHeroCraft1579, 2021-09-03:
I installed Windows 11 on my 230GB SSD but whenever i boot it takes around 10 Minutes and i get a windows error message saying "pnp driver watchdog". For installation i used the command line tool but didn't change any config. Im using a Pi 4 with 8 GB Ram

mobilegmYT, 2021-09-04:
I'll redirect you to the WoR Discord server for support: https://discord.gg/worofficial

Botspot, 2021-09-04:
I asked a WoR developer about this. He could only think of two causes for this error:

- Too high overclocking.
- Low voltage

TheHeroCraft1579, 2021-09-04:
I haven't overclocked the Pi and im using the official Power Supply that came with the Pi

TheHeroCraft1579, 2021-09-04:
I think its a power issue with my SSD because if i plug it in the USB port sometimes my keyboard doesn't respond (on another os) .

I also flashed it on my 32Gb SD Card but it always trys to boot over PXE. I also tried forcing it to boot fron SD from the setup but it just stays at the boot logo

TheHeroCraft1579, 2021-09-04:
@
[image]

TheHeroCraft1579, 2021-09-04:
Here are some screenshots showing the contents of my SD Card

[image]
[image]
[image]

Itai-Nelken, 2021-09-04:
If your ssd is connected to a USB 3 port, try a usb 2 port. If I remember correctly fast usb drives are too mucj for the usb driver. Also try powering it externaly, maybe it uses too much power?

Botspot, 2021-09-04:
> Here are some screenshots showing the contents of my SD Card
> 
> [image]
> [image]

Thank you for the screenshots. There ought to be three partitions there, and `install.wim` should have been extracted. Looks like the purple-screen installer did not finish up all the way. On first boot, did you see the purple screen with the white window in the center? 

> I think its a power issue with my SSD because if i plug it in the USB port sometimes my keyboard doesn't respond (on another os).

That could definitely be what's causing this. I used to have the same issue on Linux (SSD disconnecting) until I jerry-rigged the USB ports so the Pi's USB ports would never shut off due to over-current.

TheHeroCraft1579, 2021-09-04:
No i didn't see a purple screen.

Here's a video:
https://streamable.com/jo74nv

Botspot, 2021-09-04:
> Here's a video:
> https://streamable.com/jo74nv

Try re-flashing Windows to the SSD.

TheHeroCraft1579, 2021-09-06:
The problem is i dont have a power cable for my ssd.

Also on sdcard no success. I tried to boot manually by selecting the bootloader (efi->boot->bootaa64.efi i guess?) 
This was the result: 
[image]

TheHeroCraft1579, 2021-09-06:
I will now compare the ssd with my sdcard and see if there are different partitions, files,....

TheHeroCraft1579, 2021-09-10:
I was just impatient with the SD Card it worked after 30 Minutes👍

closed 2021-09-10 by Botspot