Botspot/wor-flasher #17: No Boot

actraiser30, 2021-09-04:
Hello

i created a installation card(32GB) in a usb card reader in latest raspbian in my Raspberry pi 3B+
everything finished well,i shutdown the system and put the card to start the installation of windows 10
but the screen does not turn on,its like i have not a card in Rpi

any suggestion?

thanks

Botspot, 2021-09-04:
Did you choose the Pi3 setting in the first window?
Did you wait 10 minutes to see if anything ever appeared on-screen? (Windows is extra slow to boot for the first time, and a Pi3 will be much slower than a Pi4)
Could you boot Raspbian on your Pi3, then insert the Windows sd card and send me screenshots of what each partition contains?

actraiser30, 2021-09-04:
yes i choose the Pi3 in the first window
i waited more than 10 minutes,the screen does not come on at all.
i also tried it in my Pi4 but got the same result

[image]

in the photo are the two visible partions in the card,the wim file is 3,6GB

Botspot, 2021-09-04:
> yes i choose the Pi3 in the first window
> i waited more than 10 minutes,the screen does not come on at all.
> i also tried it in my Pi4 but got the same result

The only solution I can think of is to delete the wor-flasher-files folder and to start over flashing. Let me know how it goes.

actraiser30, 2021-09-05:
i did and nothing changed no boot

Botspot, 2021-09-06:
> i did and nothing changed no boot

First, be aware that an image generated for a Pi3 won't work on a Pi4. And vice-versa.
Let's see if the problem is located in the Pi3-specific code in my flasher, or somewhere else.

To find out, @actraiser30, could you run WoR-flasher again as normal, but this time **select the Pi4/Pi400** setting? Then boot the flashed drive with your Pi4, of course. If the Pi4 boots, we can try to pinpoint exactly why the Pi3-specific stuff is not working. If the Pi4 does not boot, then we will have to look at your hardware setup to try to see what is wrong.

symbios24, 2021-09-06:
Pi4 boots windows 10

volodya327, 2021-09-06:
I've just got the same problem with Pi3+ and options Win10 for Pi3/Pi2: don't boot after flasing image - only black screen

symbios24, 2021-09-11:
There is also an issue on windows 10 in Rpi4 in first startup after setup ,it asks for wifi or Ethernet  internet connection to do an important update and when i press next it cannot find any connection, i have the Ethernet cable connected and there is internet,can you take a look at it?

Botspot, 2021-09-14:
> There is also an issue on windows 10 in Rpi4 in first startup after setup ,it asks for wifi or Ethernet internet connection to do an important update and when i press next it cannot find any connection, i have the Ethernet cable connected and there is internet,can you take a look at it?

Ethernet works fine for me on Windows 10/11. Perhaps this is an issue with your country's website-blocking policy, or is caused by your ISP, or maybe by your local network. If you mean that the update check didn't succeed, that's normal - Windows on Raspberry is not capable of doing OS updates the normal way. See this article: https://www.worproject.ml/guides/performing-os-updates

symbios24, 2021-09-14:
I dont mean within windows i mean in the first start after install finishes and you get a prompt to configure some things like region etc.

Botspot, 2021-09-14:
> I dont mean within windows i mean in the first start after install finishes and you get a prompt to configure some things like region etc.

If you mean that the update check didn't succeed, that's normal - Windows on Raspberry is not capable of doing OS updates the normal way. See this article: https://www.worproject.ml/guides/performing-os-updates

symbios24, 2021-09-14:
The thing is it does not go any further if it does not do the update

Botspot, 2021-09-14:
> The thing is it does not go any further if it does not do the update or check online
> I installed windows in a card not usb

I'm going to try to replicate your setup - Windows 10 on a SD card for Pi4.

Botspot, 2021-09-23:
> I'm going to try to replicate your setup - Windows 10 on a SD card for Pi4.

Results: Windows 10 worked fine for me. I used Ethernet during the installation and it proceeded smoothly on the first attempt.
I'm not sure what was going on for @symbios24, but it works fine for me.

@symbios24 is welcome to open a new issue if this problem is still occuring.

closed 2021-09-23 by Botspot