Botspot/wor-flasher #18: missing dependencies / Aria2 error

Kyliael, 2021-09-04:
Hi
I'm getting this error

`Generating Windows image with uupdump
Retrieving aria2 script...
/home/kylia/wor-flasher-files/uupdump/uup_download_linux.sh: line 55: aria2c: command not found
Failed to retrieve aria2 script

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 1 of 4)
Retrieving aria2 script...
/home/kylia/wor-flasher-files/uupdump/uup_download_linux.sh: line 55: aria2c: command not found
Failed to retrieve aria2 script

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 2 of 4)
`

Kyliael, 2021-09-04:
Note : I'm using Linux Mint20.2

Botspot, 2021-09-04:
This is very strange - WoR-flasher has a line to install aria2c when being run!

Botspot, 2021-09-04:
Perhaps the `aria2` package had been installed previously, and then you later uninstalled it? In that case, the `dpkg -l` command may detect the uninstalled package still in the cache, thereby making my script think `aria2` was already installed.

This edge-case has now been fixed.

closed 2021-09-04 by Botspot

Kyliael, 2021-09-04:
It was a fresh install

Botspot, 2021-09-04:
> It was a fresh install

In any case, I've improved the accuracy of the package-status detection. So no matter what really happened, it shouldn't happen again as my script checks for package's status using a better approach.

commit 608a0d99 (2021-09-04): fix package status detection to close #18