Botspot/wor-flasher #19: BdsDxe: No bootable option or device was found.

sanme98, 2021-09-05:
Hi, if try the wor-flasher in rpi 4 64bit OS, it has flashed successfully to the 64gb SD card, but got error during boot.

```
Error: Could not detect network connection.

Error: Could not detect network connection.
BdsDxe: No bootable option or device was found.
BdsDxe: Press any key to enter the Boot Manager Menu.
```

Have tried enter Boot Manager & select the SD/MMC on Arasan SDHCI, but after Enter, no response also.

Roburetto, 2021-09-05:
I am experiencing the same with Twister OS.
A couple of notes:
. The Pi does not have ethernet connected (is this a requirement?)
. I did not remove the Twister OS SD Card, Windows is booting from USB (BOOT_ORDER=0xf14)

sanme98, 2021-09-05:
@Roburetto Thanks for your reply. Trying to use USB SD card reader instead of plug directly to SD card port, it starting to install now :).

closed 2021-09-05 by sanme98