Botspot/wor-flasher #20: Failed to mount Error

LTstrange, 2021-09-05:
Hi, I try the wor-flasher in rpi 4b 32bit OS. I flash the win11 OS on a 7.7GB USB drive. But I meet an Error that said "Failed to mount /dev/sda2 to /media/pi/WOR-installer/winpart". Thanks for help.
[image]

Itai-Nelken, 2021-09-05:
7gb is too small. A minimum of 32gb is recommended.

LTstrange, 2021-09-05:
Thank you very much.

> 7gb is too small. A minimum of 32gb is recommended.

 by the way, is that possible to only use the sdcard that rpOS running on, and flash win11 on it self?

Itai-Nelken, 2021-09-05:
>  by the way, is that possible to only use the sdcard that rpOS running on, and flash win11 on it self? 

No

closed 2021-09-06 by LTstrange