Botspot/wor-flasher #22: Failed to determine a hashsum for WoR PE-based installer

starlord1980, 2021-09-05:
I'm getting:

Failed to determine a hashsum for WoR PE-based installer.
URL: https://worproject.ml/dldserv/worpe/gethashlatest.php

The URL is dead, any solutions?

Botspot, 2021-09-05:
This command is run: `wget -qO- https://worproject.ml/dldserv/worpe/gethashlatest.php | cut -d ':' -f2`
If the output of that command is empty, the script exits with this error:
```
Failed to determine a hashsum for WoR PE-based installer.
URL: https://worproject.ml/dldserv/worpe/gethashlatest.php
```
I'm not sure why the link is failing. Even though I was unable to reproduce your issue, I've let the website's maintainer know.
@starlord1980, what happens when you put that URL in your web browser? You said the link is "dead" - what do you mean?

starlord1980, 2021-09-05:
I dont believe it, it was my ISP blocking the URL! I connected with my phone and it worked fine. I think there must be some sort of web filtering going on, they must think it’s a dodgy website.

never mind.

Botspot, 2021-09-06:
> I dont believe it, it was my ISP blocking the URL! I connected with my phone and it worked fine. I think there must be some sort of web filtering going on, they must think it’s a dodgy website.
> 
> never mind.

Good to know. Does connecting with http work?
```
wget -O- http://worproject.ml/dldserv/worpe/gethashlatest.php
```

starlord1980, 2021-09-06:
Yes, thanks.

Botspot, 2021-09-06:
> Yes, thanks.

Fascinating. I will change the URL to use http soon.

closed 2021-09-06 by Botspot

TheDeveloper, 2021-09-17:
Using different nameservers can help with DNS issues.

On pi os, edit `/etc/dhcpcd.conf` and uncomment the line:

    #static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

and add cloudflare `1.1.1.1` into the list

    static domain_name_servers=1.1.1.1 192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

then

    sudo service dhcpcd restart

more info here https://pimylifeup.com/raspberry-pi-dns-settings/