Botspot/wor-flasher #23: Failed to generate Windows ISO

aaron-is-the-best2114, 2021-09-05:
I installed this app using Pi-apps and when I opened it I chose the OS I want, Windows 10, I chose my language, en-us, and I chose my drive, 64 GB Sandisk Ultra USB 3.0 and I let it do it's thing, but a little bit later, I got this error:

               Generating Windows image with uupdump
One of required applications is not installed.
The following applications need to be installed to use this script:
 - aria2c
 - cabextract
 - wimlib-imagex
 - chntpw
 - genisoimage or mkisofs

If you use Debian or Ubuntu you can install these using:
sudo apt-get install aria2 cabextract wimtools chntpw genisoimage

If you use Arch Linux you can install these using:
sudo pacman -S aria2 cabextract wimlib chntpw cdrtools

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 1 of 4)

Than when it got to Attempt 4 of 4, I got this error:

Failed to generate a Windows ISO! (No file named /home/pi/wor-flasher-files/uupdump/*ARM64*.ISO exists)

I am using a Raspberry pi 4 8GB model to install windows on my flash drive and I am connect VIA ethernet.  I also attached a saved copy of the terminal session that was being used to install Windows on the flash drive

[WOR - FLASHER (error).odt](https://github.com/Botspot/wor-flasher/files/7111655/WOR.-.FLASHER.error.odt)
 to this isssue as well.

Botspot, 2021-09-06:
Very interesting issue. My script was supposed to install those packages, so the fact that it's not is strange.

Botspot, 2021-09-06:
This appears to be some sort of bug in the `install_packages` function.

Botspot, 2021-09-06:
@aaron-is-the-best2114, could you try something for me?

If you installed the packages the error output recommended, **please uninstall them**. Then let's see how your system handles the `install_packages` function:
```
source ~/wor-flasher/install-wor.sh source
install_packages 'yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse exfat-utils wget udftools'
```
If ANY of the above packages are uninstalled, the `install_packages` function should install them automatically. If that does not happen, then I need to understand why.

aaron-is-the-best2114, 2021-09-06:
I will try that as soon as possible, Thanks for the help. I'll let you know if it works.

aaron-is-the-best2114, 2021-09-06:
@Botspot  It's still saying one of the packages are not installed.  
Any Ideas?

Botspot, 2021-09-06:
> @Botspot It's still saying one of the packages are not installed.
> Any Ideas?

Which package is it?

aaron-is-the-best2114, 2021-09-06:
It doesn't specify what package is not installed.
This script you gave me gave me had no errors whatsoever.    
Here's what it says when I tried to install windows again: 

Generating Windows image with uupdump
One of required applications is not installed.
The following applications need to be installed to use this script:
 - aria2c
 - cabextract
 - wimlib-imagex
 - chntpw
 - genisoimage or mkisofs

If you use Debian or Ubuntu you can install these using:
sudo apt-get install aria2 cabextract wimtools chntpw genisoimage

If you use Arch Linux you can install these using:
sudo pacman -S aria2 cabextract wimlib chntpw cdrtools

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 1 of 4)

Botspot, 2021-09-06:
I've adapted some code from the script to a debugging version that you can run in a terminal.
```
dependencies='yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse exfat-utils wget udftools'
install_list=''

PREIFS="$IFS"
IFS=' '
for package in $dependencies ;do
  if ! dpkg -s "$package" 2>/dev/null | grep -q 'Status: install ok installed' ;then
    echo "Package $package is uninstalled."
    #if the currently-checked package is not installed, add it to the list of packages to install
    if [ -z "$install_list" ];then
      install_list="$package"
    else
      install_list="$install_list $package"
    fi
  else
    echo "Package $package is already installed."
  fi
done
IFS="$PREIFS"

echo "Finished install_list: $install_list"

sudo apt install --dry-run yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse exfat-utils wget udftools

```
Please paste all of the above code into a terminal and reply with the full output.

aaron-is-the-best2114, 2021-09-06:
Here is the output: 

[output.txt](https://github.com/Botspot/wor-flasher/files/7117285/output.txt)

Botspot, 2021-09-06:
Thank you for running those debug commands.

Based on them I have figured out that it's not a package issue at all - it's an issue with the `chntpw` command.
To prove my point, run this:
```
chntpw
```
If my theory is right, that command will not work.
And yet, `chntpw` **is installed**. Your output.txt says: "`chntpw is already the newest version (1.0-1.1).`"

Someone else had this same issue on TwisterOS. The `chntpw is already the newest version (1.0-1.1).` command is located in `/usr/sbin` - an uncommon command directory.
On default Raspberry Pi OS, this isn't an issue. The `/usr/sbin` directory is included in the `$PATH` variable, meaning that when you run a command it will look in that folder.

Let's try that too. Please run this command and reply with the output:
```
echo $PATH
```

Botspot, 2021-09-06:
By the way, I just added some new lines to the script that should automatically add `/usr/sbin` to your PATH variable before running `uup_download_linux.sh`. Let me know if this ended up fixing your problem.

aaron-is-the-best2114, 2021-09-06:
So I tried it again and at first it actually started to work, but after about 2 minutes into installing windows it signed me out of the Pi to the login screen.  So when I logged back in I got this pop-up error:

[image]

So I tried it again and my screen went black as if it was logging me out as before and this time it auto logged me back in this time and I got the same pop-up error.  

Here is the  echo $PATH  results:

/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games

Botspot, 2021-09-07:
> So I tried it again and at first it actually started to work

Great! With this change, WoR-flasher now works on TwisterOS. Thanks for the help with that.

> So I tried it again and at first it actually started to work, but after about 2 minutes into installing windows it signed me out of the Pi to the login screen. So when I logged back in I got this [...]

This is not an issue with WoR-flasher, this is an issue with **your setup** and how it behaves under heavy load.
Is your Pi overclocked? If it is, turn all overclocking off and then reboot.
What power supply are you using? How often do you see the yellow lightning-bolt in the corner?
Do you monitor the Pi's USB ports? If a USB-connected device consumes too much power, the Pi's USB ports will shut off. In some circumstances, this shutting-off action will crash your system.

Botspot, 2021-09-09:
Any updates, @aaron-is-the-best2114?

symbios24, 2021-09-09:
Install wor normal and not from pi apps,
Make aure you use the latest raspbian

aaron-is-the-best2114, 2021-09-11:
Sorry, I haven't got a chance to try it again, school just started for me, so I've been busy.  I'll get back to you either  later today or tomorrow morning.

vk474, 2021-09-11:
Okay no problem

aaron-is-the-best2114, 2021-09-13:
I tried it again earlier today and I was successful.  Thanks for the help.

Botspot, 2021-09-14:
> I tried it again earlier today and I was successful. Thanks for the help.

Excellent! Enjoy Windows 10!

closed 2021-09-14 by Botspot