Botspot/wor-flasher #24: nice: '/media/pi/Sandisk/uupdump/uup_download_linux.sh': Permission denied

ITdairy, 2021-09-06:
I installed wor-flasher from pi-apps. I launch the GUI from menu/Accessories/WoR-flasher.

After configuration, the tool launch a terminal interface and everything went well till here:
```
Generating Windows image with uupdump
nice: '/media/pi/Sandisk/uupdump/uup_download_linux.sh': Permission denied

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 1 of 4)
nice: '/media/pi/Sandisk/uupdump/uup_download_linux.sh': Permission denied

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 2 of 4)
nice: '/media/pi/Sandisk/uupdump/uup_download_linux.sh': Permission denied

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 3 of 4)
nice: '/media/pi/Sandisk/uupdump/uup_download_linux.sh': Permission denied

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 4 of 4)
Failed to generate a Windows ISO! (No file named /media/pi/Sandisk/uupdump/*ARM64*.ISO exists)
```

I tried:
- Launch the GUI using `sudo /home/pi/wor-flasher/install-wor-gui.sh`
- Use CLI script with and without `sudo`
- Change line 419 in `install-wor.sh` from `nice "$(pwd)/uup_download_linux.sh"`  to `nice "sudo $(pwd)/uup_download_linux.sh"` (idk if this is a stupid change)

The issue still persists.

Here is the output (Same for both GUI and CLI, but with configuration dialog in CLI)  : [wor-flasher.log](https://github.com/Botspot/wor-flasher/files/7113070/wor-flasher.log)

Botspot, 2021-09-06:
> I installed wor-flasher from pi-apps. I launch the GUI from menu/Accessories/WoR-flasher.
>
> After configuration, the tool launch a terminal interface and everything went well till here:
>
> ``` [...]

It appears you have customized the DL_DIR variable to download everything to?

ITdairy, 2021-09-06:
> It appears you have customized the DL_DIR variable to download everything to?

Ya, I customize it to another drive as the script warn me that I don't have enough space to download the files in internal storage.

Botspot, 2021-09-06:
> Ya, I customize it to another drive as the script warn me that I don't have enough space to download the files in internal storage.

First, make sure the SanDisk drive's allows the user `pi` to read from and write to it. (Just open a file-manager without using `sudo` and see if you can create/edit a file in the drive)

Also, if your SanDisk drive is formatted with a FAT32 partition, it may not work at all because FAT32 partitions don't store file metadata the same way Ext4 partitions do. This is a possibility, and I don't know if it will work. But

My suggestion: 
Remove the `sudo`. That was a good thing to try, but in this case it won't help because the script is supposedly not marked as executable. (Even though it is. Like I said, FAT32 partitions don't store file-metadata)
Replace the `sudo` with `bash` This overrides the check-for-executable and runs the script no matter what file-permissions it has.

ITdairy, 2021-09-06:
Thx for your suggestion.
I added `bash`  in :
`install-wor.sh`: line 423, `nice "$(pwd)/uup_download_linux.sh"` to `nice bash "$(pwd)/uup_download_linux.sh"`
`uup_download_linux.sh`: line 79, `./files/convert.sh wim "$destDir" 0` to `bash ./files/convert.sh wim "$destDir" 0`

and format the SD card I want to flash Win11 to ext4.

The script runs well now.

But another problem occurs: 

[image]

Should I open a new issue for this?

Botspot, 2021-09-06:
> The script runs well now.

Good to know.

> But another problem occurs:
> 
> [image]

Interesting. This is probably due to how FAT32 partitions work. They cannot contain any file that is larger than 4GB. That is the only explanation I can think of for why the script did not detect insufficient space in DL_DIR.
Sorry for not remembering this detail earlier. Basically, your SanDisk drive will not work until it is formatted with an Ext4 partition.
To accomplish this, you can use the `gparted` tool. If you wish to retain other files on your SanDisk, you can shrink the FAT32 partition and then create a secondary Ext4 partition.

Botspot, 2021-09-06:
I've added a check for this in the script. Now, if $DL_DIR is a folder located on a FAT-type partition, the script will exit with this error:
```
The $DL_DIR directory is on a FAT32 partition. This type of partition cannot contain files larger than 4GB, however the Windows image will be 6GB.
```

ITdairy, 2021-09-07:
Thanks. However I run the script in my kali-linux and everything works well without changing any code....
I think this should be my rpi (or maybe the Sandisk drive) storage problem.

closed 2021-09-08 by Botspot

Botspot, 2021-09-09:
> Thanks. However I run the script in my kali-linux and everything works well without changing any code....
> I think this should be my rpi (or maybe the Sandisk drive) storage problem.

Does that mean this issue can be closed? Or is there something you would still like assistance with?

cycool29, 2021-09-09:
> Does that mean this issue can be closed?

Yes. Thanks

Botspot, 2021-09-09:
> Yes. Thanks

I just wanted to make sure. Thank you.