Botspot/wor-flasher #25: Debian SID - Win10 on Pi4

fiberbus, 2021-09-07:
I was unable to install using the script as written. Even though the wor-flasher gui completion screen would come up, the pi could not mount the SD card. I had to make some changes. I am not sure if they are all needed. I don't have another SD card to narrow it down.

I used exfatprogs instead of exfat-utils.
I modified the scripts. Here are the differences. YMMV.

`diff install-wor.sh my_install-wor.sh
191c191
< install_packages 'yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse exfat-utils wget udftools'
---
> install_packages 'yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse exfatprogs wget udftools'
472a473
> sudo mount.exfat-fuse "$(get_partition "$DEVICE" 2)" "$mntpnt"/winpart
474d474
< sudo mount "$(get_partition "$DEVICE" 2)" "$mntpnt"/winpart
478c478
<   sudo mount "$(get_partition "$DEVICE" 2)" "$mntpnt"/winpart || error "Failed to mount $(get_partition "$DEVICE" 2) to $mntpnt/winpart"
---
>   sudo mount.exfat-fuse "$(get_partition "$DEVICE" 2)" "$mntpnt"/winpart || error "Failed to mount $(get_partition "$DEVICE" 2) to $mntpnt/winpart"
`

`diff install-wor-gui.sh my_install-wor-gui.sh
32c32
< install_packages 'yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse exfat-utils wget'
---
> install_packages 'yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse exfatprogs wget'
`

mobilegmYT, 2021-09-07:
Currently the tool is meant for RPi OS which is on Buster, I think it might be updated to exfatprogs after RPi OS goes into bullseye too.

fiberbus, 2021-09-07:
Thank you. I posted it mainly for those who might be stuck and not know how to proceed. Best wishes.

Botspot, 2022-07-01:
This was fixed recently. Details in issue https://github.com/Botspot/wor-flasher/issues/76

closed 2022-07-01 by Botspot