Botspot/wor-flasher #28: I'm getting stuck to No bootable option or device was found

raducretu, 2021-09-07:
Trying on a Raspberry Pi4 8Gb

If I unplug the ethernet cable 
[https://www.dropbox.com/s/6as3k0zbmr1b844/IMG_6650.JPG?dl=0](https://www.dropbox.com/s/6as3k0zbmr1b844/IMG_6650.JPG?dl=0)

If I put back the ethernet cable
[https://www.dropbox.com/s/k4i8b8x9a767mxl/IMG_6649.JPG?dl=0](https://www.dropbox.com/s/k4i8b8x9a767mxl/IMG_6649.JPG?dl=0)

AndryX19, 2021-09-07:
Same error for me (i'm on pi4 4gb with usb 3.0 sata ssd, all things updated). On first boot, after finished loading drivers and go for installation process, it's start searching for installation disk but it end with no installation disk was found, and also mouse and keyboard stop working.
I tried to doing the installation from windows with pc, and after making the ssd bootable, when going to start raspberry it faces inacessible boot device bsod.... :(

Botspot, 2021-09-08:
Errors like this were caused by the Pi4 having an outdated EEPROM.
After some troubleshooting with others on Discord, we've worked out two solutions:

- Update the EEPROM.
- Use [an older version](https://github.com/pftf/RPi4/releases/tag/v1.28) of the UEFI.

This issue should be solved, as WoR-flasher [now downloads](https://github.com/Botspot/wor-flasher/commit/28a22ab484fb0871f5186d49d16a327804e467fb) an older UEFI package to accommodate these outdated Piess.

Note: very-early Pi4s won't run WoR at all until the EEPROM is updated, so consider following [these instructions](https://www.tomshardware.com/how-to/boot-raspberry-pi-4-usb) if you still encounter booting problems.

AndryX19, 2021-09-08:
As I write before, I update all things, included bootloader....

Botspot, 2021-09-08:
> As I write before, I update all things, included bootloader....

Please delete the `~/wor-flasher-files/uefipackage` folder and proceed through the flashing process again. By deleting the folder, WoR-flasher will re-download the fixed UEFI package.

azhuge233, 2021-09-08:
Same thing happens to me.

>Same error for me (i'm on pi4 4gb with usb 3.0 sata ssd, all things updated). On first boot, after finished loading drivers and go for installation process, it's start searching for installation disk but it end with no installation disk was found, and also mouse and keyboard stop working.

I tried changing [this URL variable](https://github.com/Botspot/wor-flasher/commit/28a22ab484fb0871f5186d49d16a327804e467fb#diff-d99d6dcec506a8352af9322c4e6a1db3622e63a31f2b1b0ba27731130f381b8bR411) to the latest [pftf/RPi4](https://github.com/pftf/RPi4) version, which is v1.30, the drive can be detected and the installation can continue.

Botspot, 2021-09-08:
> I tried changing the UEFI firmware download string to the latest [pftf/RPi4](https://github.com/pftf/RPi4) version, which is v1.30, the drive can be detected and the installation can continue.

@azhuge233 Were you trying to install Windows 10 or Windows 11?

azhuge233, 2021-09-08:
@Botspot Windows 11 on RPi 4 8GB version.

Botspot, 2021-09-09:
It's good this problem has been solved for you. Enjoy Windows 11!

closed 2021-09-09 by Botspot

AndryX19, 2021-09-09:
thanks guys, i'll try today this fix and report

Waymorehoss, 2024-07-24:
I need some help please. i am trying to put windows 10 on my pi but when i boot it shows a network error i can't understand what @Botspot is talking about . i created a seperate partition and put the firmware files in it so what do i do now just run the wor installer ? i'm on a windows machine so i could not find the  ~/wor-flasher-files/uefipackage directory.