Botspot/wor-flasher #30: Error: Could not detect network connection.

ITdairy, 2021-09-08:
I don't know if this is wor-flasher problem or the UEFI. 
When I boot the sd card, it shows 
[image]
Any fix?

shubhamkastwar1, 2021-09-08:
Facing same issue "pxe over ipv"
I know i need to disable boot from network and disable secure boot but not able to find it in bios.

Also i am testing it with sd card and in bios it is showing emmc..
Any idea how to fix

linus-oof-tips, 2021-09-08:
I had that to I reinstalled

Botspot, 2021-09-08:
This problem is caused by your Pi4 still using an outdated version of the bootloader EEPROM.

To fix it, WoR-flasher now uses an older version of UEFI to accommodated outdated Pies.
To apply the fix:

- Make sure wor-flasher is up to date,
- Then delete this folder so that the older version of UEFI is downloaded: `~/wor-flasher-files/uefipackage`
- Flash the drive again. No need to re-generate the image.

shubhamkastwar1, 2021-09-08:
Now i am getting this error 
[image]

shubhamkastwar1, 2021-09-09:
[image]

Botspot, 2021-09-09:
I accidentally forgot to change a set of single-quotes to double-quotes in the URL. Sorry about that. It was [fixed](url) within minutes, and so should work fine now.

shubhamkastwar1, 2021-09-09:
I am trying now, Let's hope it work's. :)

shubhamkastwar1, 2021-09-09:
[image]
I think this is working now

shubhamkastwar1, 2021-09-09:
@Botspot thanks for the help :)

Botspot, 2021-09-09:
> @Botspot thanks for the help :)

No problem! It's good to know this is now working for you.

closed 2021-09-09 by Botspot

vk474, 2021-09-09:
Do not west your time friends because that  pettern I have done for time
and results came same  show it wile not work 😌

Botspot, 2021-09-09:
> Do not west your time friends because that pettern I have done for time and results came same show it wile not work 😌

@vk474, if you are still having an issue with WoR-flasher, please [open a new issue](https://github.com/Botspot/wor-flasher/issues/new/choose).

cycool29, 2021-09-09:
Thanksss now it worked and i installed windows 11 on rpi successfully!
[image]

Alfpv, 2023-11-01:
Hey, I am getting the same 2 'could not detect network connection' errors, but none of the others, is that ok? Or should I be worrying?

Botspot, 2023-11-02:
> Hey, I am getting the same 2 'could not detect network connection' errors, but none of the others, is that ok? Or should I be worrying?

That is fine.

Subhojit131, 2024-11-25:
> > Hey, I am getting the same 2 'could not detect network connection' errors, but none of the others, is that ok? Or should I be worrying?
> 
> That is fine.

I am also getting the 2 could not detect network connection errors and its not getting past that screen I have waited for 30 minutes and more still its stuck

not-real-alex, 2024-11-25:
This issue will solve itself with time. if not 

this is caused by a slow boot drive "SD card or USB Drive"
you could have this same issue wired to the internet or not connected at all.
Sometimes reinstalling the operating system to the drive helps.

Subhojit131, 2024-11-26:
> This issue will solve itself with time. if not 
> 
> this is caused by a slow boot drive "SD card or USB Drive"
> you could have this same issue wired to the internet or not connected at all.
> Sometimes reinstalling the operating system to the drive helps.  

I also thought that this could be possible thats why I tried changing the drive but even after changing it is stuck in that same screen could it be related to the EPROM? I will try to update it and try again today