Botspot/wor-flasher #31: WoR-PE

linus-oof-tips, 2021-09-08:
It says it has not disk and my keyboard and mouse is not working

Botspot, 2021-09-08:
> It says it has not disk and my keyboard and mouse is not working

Your mouse and keyboard should be working. Check and make sure that your USB devices are receiving **power** through the USB ports.

linus-oof-tips, 2021-09-08:
The USB ports does not work but it works is the bios so I don’t know the problem

Botspot, 2021-09-09:
The USB ports does not work but it works is the bios so I don’t know the problem

The "usb ports not working" issue you are having sounds exactly similar to [this person's issue](https://github.com/Botspot/wor-flasher/issues/32).
They fixed it by deleting the `~/wor-flasher-files` folder and re-running WoR-flasher. See if this helps you too.

Botspot, 2021-09-09:
Any updates, @linus-oof-tips?

symbios24, 2021-09-09:
It looks like this guy cannot explain a thing

AndryX19, 2021-09-10:
Hi, disk error still there.
[image]
[image]
[image]

linus-oof-tips, 2021-09-10:
> Any updates, @linus-oof-tips?

Um not right now I’m gonna try again today

Botspot, 2021-09-10:
> Hi, disk error still there. Also for me unusable mouse and keyboard after boot, but keyboard work in bios

@AndryX19, please [open a new issue](https://github.com/Botspot/wor-flasher/issues/new) so that your problem can be solved separately.

Botspot, 2021-09-14:
> > Any updates, @linus-oof-tips?
> 
> Um not right now I’m gonna try again later today

How is progress going @linus-oof-tips?

linus-oof-tips, 2021-09-14:
> > > Any updates, @linus-oof-tips?
> > 
> > 
> > Um not right now I’m gonna try again later today
> 
> How is progress going @linus-oof-tips?

I got a lot of school work right now so soon when I have time

Sussy-OS, 2022-02-28:
> > > > Any updates, @linus-oof-tips?
> > > 
> > > 
> > > Um not right now I’m gonna try again later today
> > 
> > 
> > How is progress going @linus-oof-tips?
> 
> I got a lot of school work right now so soon when I have time

While ago

WombleWoo7547, 2022-06-25:
Are you using USB Boot? I haven't tried it but it might fix the problem

Botspot, 2022-07-01:
Closing this issue due to inactivity.

closed 2022-07-01 by Botspot