Botspot/wor-flasher #32: Couldn't find installation disk

mboroff, 2021-09-08:
I am using WorPi flasher using the manual download and run procedure. After doing the equivalent bios setting and rebooting, the Windows install dies after loading the drivers. I get a pop-up that states "Couldn't find the installation disk!". I can not check the logs. This occurs whether it is Windows 10 pro or Windows 11 pro.

Botspot, 2021-09-08:
Please update your Pi's bootloader EEPROM and try again.
If it still does not find the disk, try changing this line in the `install-wor.sh` script from this:
```
URL="https://github.com/pftf/RPi${RPI_MODEL}/releases/download/v1.28/RPi${RPI_MODEL}_UEFI_Firmware_v1.28.zip"
```
To this:
```
URL="https://github.com/pftf/RPi${RPI_MODEL}/releases/download/v1.30/RPi${RPI_MODEL}_UEFI_Firmware_v1.30.zip"
```
And then run through the WoR-flasher process again.

Changing this URL will use a newer version of the UEFI (similar to a BIOS). Using UEFI v1.30 works better for some people, but for other people it prevents their Pi from booting.
If updating the EEPROM did not solve the problem but using UEFI v1.30 did solve the problem, I will have to look into a new solution to make this work for everyone, regardless of what version bootloader their Pi4 ships with.

mboroff, 2021-09-08:
It still fails. Are there logs I can find on the ssd?

I am attaching a screen shot of the fail and shot of my update and eprom results and the lines in the install.sh.

[image]

[image]

[image]

Botspot, 2021-09-08:
> It still fails. Are there logs I can find on the ssd?

Logs can be viewed by clicking the "Logs" button in the top-right corner of the white window.

> [image]

That looks correct.

> [image]

Please run this command:
```sudo rpi-eeprom-update -a```

mboroff, 2021-09-08:
ok. Now run the install again?
[image]

Botspot, 2021-09-08:
> ok. Now run the install again?
> [image]

Reboot Linux. Then shutdown Linux. Then boot Windows and see if it works. Hopefully you won't need to re-flash again.

mboroff, 2021-09-08:
Same result. Can't the installation disk.

Botspot, 2021-09-08:
> Same result. Can't the installation disk.

Okay, please view the logs. Logs are accessible by clicking the Logs button.
I assume you re-flashed Windows 11, after changing the script to use UEFI v1.30?

mboroff, 2021-09-08:
Yes it was windows 11 I tried to install. I can not click on anything. I have no cursor. I can't tab or just press the enter key. It seems to be stuck. Can I get to the logs if I boot up under Debian?

Botspot, 2021-09-08:
> Can I get to the logs if I boot up under Debian?

I have no idea.

> Yes it was windows 11 I tried to install. I can not click on anything. I have no cursor. I can't tab or just press the enter key. It seems to be stuck.

Your mouse ought to be able to move. I can only think of two explanations for this:

1. Your mouse and keyboard are abnormal and were not detected.
2. Your usb ports have been turned off due to overcurrent. This can happen when booting from a power-hungry SSD and not providing it with its own dedicated power source, like a powered USB hub for example.
3. The first time you ran WoR-flasher, it failed to fully extract the drivers. The script would have exited with an error, but it would have left behind an empty folder at `~/wor-flasher-files/driverpackage`. Subsequent runs would have seen that folder and moved on, effectively breaking the resulting image.

To rule out any sort of past failed extraction/download problems, please delete the entire `~/wor-flasher-files` folder and run through the flashing process again. Hopefully this is the last time!🤞

mboroff, 2021-09-08:
My keyboard and mouse are the official Raspberry ones. The mouse plugs into the keyboard. Moving the mouse directly to the usb 2 on the pi did not help. I can try a different keyboard and mouse. When I download a fresh copy should I modify the install.sh to.30?

Botspot, 2021-09-08:
> When I download a fresh copy should I modify the install.sh to.30?

You can - it shouldn't hurt as your Pi4 now uses the latest bootloader. (old uefi version was only necessary for outdated-bootloader pi4s)

You can delete and re-download the `~/wor-flasher` folder if you want, but the important folder to delete is `~/wor-flasher-files`.

mboroff, 2021-09-08:
Much better results. I am into the install right now. I decided to replace the Pi with a 2Gb Pi 4b. I completely re-did Debian, did the full upgrade, checked the eprom, down loaded the flasher, modified the script to 1.30. Omn the reboot I only changed the resolution to 720 and then let it go. If it comes up on the 2Gb pi I will move the ssd to the 4Gb and open the memory.  If you like I can close the issue and if I hit another problem it will be another issue to open. Thank you so much for your help. Patience is a virtue.

Botspot, 2021-09-08:
> Much better results.

Excellent!

> Patience is a virtue.

It sure is. Thank you for your time and patience. As a result, WoR-flasher will soon be more reliable than ever: I will soon update it to prevent the issue that occurred for you: a failed extraction. Now, if extracting a component fails, the extract-to folder will be deleted before the script exits.

closed 2021-09-08 by Botspot

Butteritron, 2023-09-02:
For me, the solution was surprisingly simple: all i had to do was plug the sd reader into a different usb port