Botspot/wor-flasher #33: Incomplete install "Insufficient system resources exist to complete the requested service"  on Rpi3B+ for both Win 10 and 11.

dgarra, 2021-09-10:
1) Excellent effort and work here. 

I think this is windows itself, but is there soft threshold on like core count or memory before windows bombs out? This only has 1GB of RAM, so not shocking, but hey, the option is there....

This seems to fail in the same spot on both versions 10 and 11 when selecting the Rpi3B+ and leaving everything default. 

I'm writing to a 29GB SD card. Runs four about an hour before dying citing resources, same failure 5 times in a row. 

```
2021-09-04 04:07:53.7657 [info] WoR-PE (version 1.0.0.0)
2021-09-04 04:07:53.9911 [info] Copyright © 2018-2021 Mario Bălănică. All Rights Reserved.
2021-09-04 04:07:54.0278 [info] OS: Windows (TM) 10 Preinstallation Environment ver 2009 (22454.1000.arm64fre.rs_prerelease.210903-1516)
2021-09-04 04:07:54.1288 [info] Loading driver list...
2021-09-04 04:07:54.2450 [info] Loading driver (1/3): mcci_dwchsotg\mcci_dwchsotg_hcd.inf
2021-09-04 04:07:54.3303 [debug] Launch pnputil.exe with the following arguments: /add-driver "X:\drivers\mcci_dwchsotg\mcci_dwchsotg_hcd.inf" /install
2021-09-04 04:07:56.1879 [debug] pnputil.exe process output:
2021-09-04 04:07:56.1879 [debug] Standard output:
Microsoft PnP Utility

Adding driver package:  mcci_dwchsotg_hcd.inf
Driver package added successfully.
Published Name:         oem0.inf
Driver package installed on device: ACPI\BCM2848\0

Total driver packages:  1
Added driver packages:  1

2021-09-04 04:07:56.1879 [debug] Exit code: 0
2021-09-04 04:07:56.2483 [info] Loading driver (2/3): mcci_dwchsotg\mcci_dwchsotg_hub.inf
2021-09-04 04:07:56.2483 [debug] Launch pnputil.exe with the following arguments: /add-driver "X:\drivers\mcci_dwchsotg\mcci_dwchsotg_hub.inf" /install
2021-09-04 04:07:57.3778 [debug] pnputil.exe process output:
2021-09-04 04:07:57.3778 [debug] Standard output:
Microsoft PnP Utility

Adding driver package:  mcci_dwchsotg_hub.inf
Driver package added successfully.
Published Name:         oem1.inf
Driver package installed on device: USB\VID_040E&PID_0100&McciDwcHsOtgHub\0

Total driver packages:  1
Added driver packages:  1

2021-09-04 04:07:57.3778 [debug] Exit code: 0
2021-09-04 04:07:57.4088 [info] Loading driver (3/3): rpisdhc\rpisdhc.inf
2021-09-04 04:07:57.4234 [debug] Launch pnputil.exe with the following arguments: /add-driver "X:\drivers\rpisdhc\rpisdhc.inf" /install
2021-09-04 04:07:58.5118 [debug] pnputil.exe process output:
2021-09-04 04:07:58.5118 [debug] Standard output:
Microsoft PnP Utility

Adding driver package:  rpisdhc.inf
Driver package added successfully.
Published Name:         oem2.inf
Driver package installed on device: ACPI\BCM2855\0

Total driver packages:  1
Added driver packages:  1

2021-09-04 04:07:58.5118 [debug] Exit code: 0
2021-09-04 04:07:58.5491 [info] Drivers loaded successfully!
2021-09-04 04:08:02.9663 [info] Loading disks...
2021-09-04 04:08:05.1935 [info] Disk 0 - Generic Mass-Storage USB Device - Removable Media - 29 GB
2021-09-04 04:08:05.1935 [info] 1 disk(s) found
2021-09-04 04:08:05.2152 [info] Finding PE disk...
2021-09-04 04:08:05.2541 [info] Mounting partitions on disk 0...
2021-09-04 04:08:05.2815 [debug] Creating diskpart script with the following commands: select disk 0 , select partition 1 , assign letter=A , exit
2021-09-04 04:08:05.3381 [debug] Launch diskpart.exe with the following arguments: /s "X:\diskpart.script"
2021-09-04 04:08:48.2117 [debug] diskpart.exe process output:
2021-09-04 04:08:48.2117 [debug] Standard output:

Microsoft DiskPart version 10.0.22454.1000
[... 308 lines cut ...]

Partition 4 is now the selected partition.

DiskPart successfully assigned the drive letter or mount point.

Leaving DiskPart...

2021-09-04 04:18:29.9096 [debug] Exit code: 0
2021-09-04 04:18:29.9608 [info] Windows partition created! (letter: B)
2021-09-04 04:18:29.9889 [info] Applying the WIM image to: B
2021-09-04 04:43:10.6482 [error] System.ComponentModel.Win32Exception (1450): Insufficient system resources exist to complete the requested service.
   at Microsoft.Wim.WimgApi.ApplyImage(WimHandle imageHandle, String path, WimApplyImageOptions options) in Microsoft.Wim.dll:token 0x6000020+0x3a
   at WoR_PE.Wrappers.WIMGAPIWrapper.<>c__DisplayClass6_0.<ApplyImage>b__0() in WoR-PE.dll:token 0x60001c1+0x52
   at System.Threading.Tasks.Task.InnerInvoke() in System.Private.CoreLib.dll:token 0x6002d62+0xf
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state) in System.Private.CoreLib.dll:token 0x60029bb+0x4c
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread) in System.Private.CoreLib.dll:token 0x6002d61+0x12e
--- End of stack trace from previous location ---
   at WoR_PE.Wrappers.WIMGAPIWrapper.ApplyImage(String imageFile, String applyPath, UInt32 imageIndex, ProgressCallback callback) in WoR-PE.dll:token 0x6000084+0x9a
   at WoR_PE.Setup.SetupProcess.ApplyImageTask() in WoR-PE.dll:token 0x60000e3+0xe8
   at WoR_PE.Setup.SetupProcess.DeploymentTask() in WoR-PE.dll:token 0x60000de+0x29b
   at WoR_PE.Setup.SetupProcess.Run(IWin32Window parent) in WoR-PE.dll:token 0x60000dc+0xa7
2021-09-04 04:43:11.1954 [warn] The installation could not be finished!
2021-09-04 04:50:17.8048 [info] Elapsed time: 00:41:07
```

Botspot, 2021-09-11:
I forwarded this to the developer of the PE-based installer. He says he thinks it's due to low RAM and will look into fixing this.

wzrd-1, 2022-07-10:
I ran into an out of memory fault in BCM2711 on Pi4, running Raspbian 11. It reset my X session during the Microsoft download phase, leaving me with everything closed at the window manager login screen.
It turned out to be caused by a previous experiment in allocating 512 to the GPU. Once I reduced it to 256, everything worked properly.
Of course, complicating the issue was, I was certain that I had reverted that change.

closed 2024-02-10 by Botspot