Botspot/wor-flasher #34: Couldn't find installation disk

AndryX19, 2021-09-11:
I repeat the process with wor-flasher from raspbian, also with the updated uefi firmware (1.3.1), same error..
[image]
[image]
[image]

Botspot, 2021-09-11:
Try updating the EEPROM on the Pi4.

From Raspberry Pi OS:
```
sudo rpi-eeprom-update -a
```

AndryX19, 2021-09-11:
it's just the latest

Botspot, 2021-09-14:
> it's just the latest

@AndryX19 What do the logs look like?

AndryX19, 2021-09-15:
forgive the wait, I had exams at the university, tomorrow I send the logs , I hope we can solve it

AndryX19, 2021-09-16:
@Botspot where are locate the log files?

Botspot, 2021-09-16:
> @Botspot where are locate the log files?

You need to click the "Logs" button. If you didn't, then try booting the pi again and hoping the same error occurs.

AndryX19, 2021-09-16:
I can't use mouse and keyboard after boot and I can't click on logs button...

AndryX19, 2021-09-19:
@Botspot I have news. Investigating I discovered that the problem lies in the bootloader and its inability to read sectors of 4096 compared to the classic 512. Since the external sata-usb supports emulate the passage from 4096bit to 512bit and therefore extend the compatibility to old systems and limitations, as in this case. So I'll try another older sata-usb adapter and hope it works, because with other systems it works but with win doesn't

christopherdev, 2021-09-22:
@AndryX19 I have the same issue, I used an old enclosure and a new sata adapter and both shows the same error and can't view logs. What sata adapter that is recommended for this to work?

MuffinGod1, 2021-09-22:
I also have the same issue. Couldn't read install disk, keyboard and mouse unresponsive. Trying to install in an argon 1 m.2 enclosure.

AndryX19, 2021-09-22:
I'm waiting for deskpi pro v2, same but also for 2.5" sata disk (in my case ssd) and m.2 disk

zShqdows, 2022-02-15:
Should this be closed due to inactivity @Botspot ?

closed 2022-02-15 by Botspot