Botspot/wor-flasher #35: Failed to mount ISO file

mortlo, 2021-09-12:
RPi 400, new install on 32 gb dd card, with apt updates, with external SSD sandisk.

went thru the whole procedure smoothly until ‘ mounting the image’
Trying again after loading the ‘udf’ kernel module. 
modprobe: error ..could not open moddep file. 
udf not found in directory. 
mount: ..unknown file system type udf.

Botspot, 2021-09-12:
> RPi 400, new install on 32 gb dd card, with apt updates, with external SSD sandisk.

Exactly what operating system? If it's Raspberry Pi OS, what variant is it? (Lite, Desktop, Full)

Maybe try installing the `raspberrypi-kernel-headers` package?

mortlo, 2021-09-13:
I used the 'raspberry Pi Imager' on its default setting of desktop (without recommended software.)
Which variant is preferable? I can easily redo it again.

is that a simple task of "sudo apt-get raspberrypi-kernel-headers"
..I'll give that a try.

RUHavingFun, 2021-09-13:
Same issue - retrying now.  Used a 64 GB SD with Raspbian 32bit standard desktop fully updated and a 128 GB SSD.  Will try again after `sudo apt-get install raspberrypi-kernel-headers`

RUHavingFun, 2021-09-13:
Failed to mount ISO file (/home/pi/wor-flasher-files/uupdump/22454.1000_PROFESSIONAL_ARM64_EN-US.ISO) to /home/pi/wor-flasher-files/isomount

Botspot, 2021-09-13:
> Trying again after loading the ‘udf’ kernel module.
> modprobe: error ..could not open moddep file.
> udf not found in directory.
> mount: ..unknown file system type udf.

Try this:
```
sudo modprobe udf
```

RUHavingFun, 2021-09-13:
`sudo modprobe udf`

modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.10.17-v7l+/modules.dep.bin'
modprobe: FATAL: Module udf not found in directory /lib/modules/5.10.17-v7l+

Botspot, 2021-09-13:
> could not open moddep file '/lib/modules/5.10.17-v7l+/modules.dep.bin'

This is a problem with your system. Please read [this forum thread](https://www.raspberrypi.org/forums/viewtopic.php?t=262963) and try the solutions in there.

RUHavingFun, 2021-09-13:
I will go see what that sheds light on.  However, I will say that I built this 32 bit Raspbian SD card from scratch last night specifically for this process.  I then did the `sudo apt update && sudo apt full-upgrade`.  I did use the advanced menu to set the SSH and WiFi.  I will do it again from scratch and not do that and check.  I also used a 64 GB SD card as it kept saying my 16 GB card was not big enough to run.

RUHavingFun, 2021-09-13:
That fixed it!  Now it completed to the "Next Steps" window.  Time to let it do that part now.

Botspot, 2021-09-13:
> I will go see what that sheds light on. However, I will say that I built this 32 bit Raspbian SD card from scratch last night specifically for this process. I then did the `sudo apt update && sudo [...]

I just figured out a plausible explanation for the issue you're having. You upgraded your packages (which included a kernel upgrade), but **you did not reboot your Pi** after the upgrades.
This, in effect prevented your Pi from finding any kernel modules. Your system was looking for modules for the current kernel (5.10.17-v7l+), but that folder was renamed during the upgrade. (5.10.60-v7l+)

mortlo, 2021-09-14:
those are exactly the same steps that I did.. too keen to see win11 working to think about rebooting - Grrr!
When youre dealing with wot seems impossible sometimes the obvious gets over looked. 

..so to make sure Im understanding correctly - all I need to do is reboot..and repeat the process..??

Botspot, 2021-09-14:
> those are exactly the same steps that I did.. too keen to see win11 working to think about rebooting - Grrr!
> When youre dealing with wot seems impossible sometimes the obvious gets over looked.
> 
> ..so to make sure Im understanding correctly - all I need to do is reboot..and repeat the process..??

What I mean is: if you installed Raspberry Pi OS on a SD card, did the apt update and upgrade like a good Linux citizen, **and then tried using WoR-Flasher without first rebooting**, the flashing process will fail due a mis-match in the running kernel versus the newly-installed kernel.
As I said before, if you get this "modprobe udf" error, try rebooting your Pi.

Of course, if you've already gotten Windows running on the Pi, then disregard everything I said above.

mortlo, 2021-09-14:
yes - I understood that I should have rebooted before commencing with WoR-Flasher.
Ive applied the fix as stated by RUHavingFun, went ok and rebooted...back into the RPi system.
This is a new issue I think...due to the SSD drive perhaps.  Ive had issues with getting other op sys booting from. 

I consider the problem I raised as fix...so thanks to all!!
..and am still amazed win11 on a RPi is possible!!!

closed 2021-09-14 by mortlo

Botspot, 2021-09-14:
To prevent this mystery from repeating itself, I've [added these lines](https://github.com/Botspot/wor-flasher/commit/3456627693bdc755cb4081c585d9f7807f41b913) of code to the install-wor.sh:
```bash
#Make sure modules for running kernel exist - otherwise a kernel upgrade occurred and the user needs to reboot. See https://github.com/Botspot/wor-flasher/issues/35
if [ ! -d /lib/modules/$(uname -r) ];then
  error "The running kernel ($(uname -r)) does not match any directory in /lib/modules.
Usually this means you have not yet rebooted since upgrading the kernel.
Try rebooting.
If this error persists, contact Botspot - the WoR-flasher developer."
fi
```

mortlo, 2021-09-14:
Excellent!!

as an additional bit of info.. I ran the flasher from scratch once again... (after a reboot)
and apart form an error message that looked like it had failed due to my SSD, on reboot it WORK!!!