Botspot/wor-flasher #36: Unkown value for RPI_MODEL. Expected  '3' or '4'.

TenBryBry2003, 2021-09-12:
I'm trying to flash windows 10 for raspberry pi 3 B onto a 129 gb san disk pro extreme micro SD card. Each time I attempt to flash my SD card, I get this error. If it helps, I am running WoR flasher from a raspberry Pi 3 B.

TenBryBry2003, 2021-09-12:
I also tried flashing windows 11, but I got the same error.

dgarra, 2021-09-12:
I ran into this myself only starting today. I reimaged my entire OS thinking something blew up, but after repopulating, still the same thing....Even tried forcing the variable. 

Worked on this system two days ago, so I'm baffled. Seeing as someone else is reporting it tho, I feel a little better. 

I went through the script, I can't tell exactly what OS or system variable may be getting called here for it to fail. 

It's 100% a 3b+. 

[image]

Botspot, 2021-09-13:
Thank you for opening an issue. This issue has been solved with [this change](https://github.com/Botspot/wor-flasher/commit/728dd18c0f1dd03b4033553b1d5667fa71519932) I just made to the script.

The problem was caused when I changed the value for the Pi3 setting in one place, and forgot to change it in the other place. As a result, the variable was not being set correctly.

closed 2021-09-13 by Botspot