Botspot/wor-flasher #38: Could not detect network device

TheTown777, 2021-09-14:
Im trying to install Windows 11 from my rpi4 but at the bios it said could not Could not detect network device. It was booted off of a sd card and i installed it from my external hdd and i need windows on the sd card So i can fix my laptop

Botspot, 2021-09-14:
When did you first install wor-flasher on a linux computer?
Could you take a picture of your screen? When was the last time you updated your Pi's bootloader?

TheTown777, 2021-09-14:
Hey im sorry i frogot i made this and i keep on having issues then later it working but my mom said she is going to get another usb drive but everytime i try to boot it from usb the bios doesent show or it does that

TheTown777, 2021-09-14:
And my laptop got a virus and destroyed the bootloader and i dont have another pc to fix it i only had a pi and the external hdd is a seagate 450gig drive

TheTown777, 2021-09-14:
Its only working on usb drives for some reason so im going to go to the store and buy another usb drive so i can use rufus but thanks so much for making this

Botspot, 2021-09-16:
> Its only working on usb drives for some reason so im going to go to the store and buy another usb drive so i can use rufus but thanks so much for making this

Does that mean this issue can be closed?

TheTown777, 2021-09-16:
> > Its only **working on usb drives** for some reason so im going to go to the store and buy another usb drive so i can use rufus but **thanks so much** for making this
> 
> 
> 
> Does that mean this issue can be closed?

Yes

closed 2022-07-01 by Botspot