Botspot/wor-flasher #40: "The Installation has failed" error

ankur1016, 2021-09-15:
I am able to create a Win11 image on the SD card, but when I boot pi with the Win image card, I get a "The Installation has failed" error" right around 40%. 

Can someone please help me with this issue and let me know what I am doing wrong?

[image]

Botspot, 2021-09-15:
The error your encountered occurred when using the Windows Preinstallation Environment. I did not make that and I don;'t know much about how it works.

[This chat group](https://discord.gg/worofficial) does contain the developers who made the PE-installer - I suggest you ask them for help. (His Discord username is @mariob)

Botspot, 2021-09-15:
I've asked mariob on your behalf. He saw the "Elapsed time" was 18 hours and thinks the most likely cause for this error is that you're using a trash SD card.
Try a high-speed, A1-rated, genuine, SanDisk or Samsung SD card that's at least 32GB.

ankur1016, 2021-09-15:
Thank you for your help.. I was not able to get in touch with MarioB. I am using Samsung 32GB EVO Plus UHS-1 card. Maybe I need UHS-3 or try a different brand of card and try. Thanks again for your help.

ankur1016, 2021-09-16:
After buying new SanDisk 32GB Extreme microSDHC, installation was complete. 

https://www.amazon.com/SanDisk-Extreme-microSDXC-Memory-Adapter/dp/B06XWMQ81P?ref_=ast_sto_dp&th=1#

Botspot, 2021-09-16:
> After buying the new SanDisk 32GB Extreme microSDHC, the installation was complete.
> 
> Thank you for all your help.

Excellent! Thank you for your patience and I hope you enjoy Windows 11.
(I may add a speed test to weed out bad sd cards sooner)

closed 2021-09-16 by Botspot