Botspot/wor-flasher #41: Errors when downloading Windows 11

SC13NC3F1CT10N, 2021-09-21:
I ran the script without modification on my Raspberry Pi 4 8GB model. It seems to go well until it stops with this message:

"Exporting Windows 10 Professional to install.wim...
[ERROR] "/Windows/System32/CatRoot/{F750E6C3-38EE-11D1-85E5-00C04FC295EE}/Microsoft-Windows-Enterprise-Desktop-Shared-Removable-Package~31bf3856ad364e35~arm64~~10.0.22458.1000.cat": blob not found
        SHA-1 message digest of missing blob:
        2463db131bcacb290589f826f687d4e29b7bc02a
ERROR: If this is a delta WIM, use the --ref argument to specify the WIM(s) on which it is based.
ERROR: Exiting with error code 55:
       A file resource needed to complete the operation was missing from the WIM.
Failed to export Windows 10 Professional to install.wim

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 4 of 4)
Failed to generate a Windows ISO! uup_download_linux.sh exited with an error so please see the errors above."

I have attempted to do this multiple times even clearing out the uupdump folder. Can anyone offer a solution?

Botspot, 2021-09-22:
This appears to be a momentary issue with [uupdump](https://uupdump.net), as a new Windows update is rolling in.

You have two options:

1. Wait for the problem to solve itself.
2. Use an older Windows-update ID.

How to find an older Windows update ID:

- go to https://uupdump.net, scroll down to **Recently added builds**.
- find  an entry that is **arm64**.
- Copy that line's **Update ID**.

Here's one I picked: "Cumulative Update for Windows 11 (22000.194) arm64" - `d0a7e04d-5f20-4cf8-97ee-895fe957174d`
In WoR-flasher, choose "**Custom**" and paste in `d0a7e04d-5f20-4cf8-97ee-895fe957174d`.

SC13NC3F1CT10N, 2021-09-22:
I'll try that. Thanks for the tip.

KAYFLUHR, 2021-09-22:
Hi there, I had the same problem and it worked with above UUID. Furthermore my SD card did not boot. What helped was to re-run the patch but BEFORE I updated my Raspi4 :
sudo apt update
sudo apt upgrade
sudo rpi-update
reboot
sudo rpi-eeprom-update -d – a
reboot

closed 2021-09-23 by Botspot

Botspot, 2021-09-23:
For now, I've changed WoR-flasher to use the older update-ID when you select Windows 11. It will remain this way until UUPdump fixes their site.

unksoldr, 2022-09-17:
Old versioin doesn't work for me same issue

commit 9915f491 (2021-09-23): install-wor.sh: use older Update-ID to close #41