Botspot/wor-flasher #43: WOR Flasher errors when i try to create Windows 11 on 250GB SSD

ejwjohn, 2021-09-23:
I am no expert on Linux , but i have tried your script and it errors shortly after starting,

The script acknowledges that there is enough storage prior to starting the build

The error indicates that there is insufficient storage to create the Windows image???, but the selected drive is far larger than i need, i get the same issue if i try to load image onto a USB stick 32Gbyte free.

Thanks
JohnW

Botspot, 2021-09-23:
> The script acknowledges that there is enough storage prior to starting the build
> 
> The error indicates that there is insufficient storage to create the Windows image???, but the selected drive is far larger than i need, i get the same issue if i try to load image onto a USB stick 32Gbyte free.

Exactly what error do you get?

ejwjohn, 2021-09-23:
> Botspot ***@***.***>
> 23 September 2021 at 20:54
>
> Exactly what error do you get?

Hello,

The following is the text generated by the WOR installation script after 
completing the initial question/Answers

Input configuration:
DL_DIR: /home/pi/wor-flasher-files
RUN_MODE: gui
RPI_MODEL: 4
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 1
UUID: d0a7e04d-5f20-4cf8-97ee-895fe957174d
WIN_LANG: en-gb
CONFIG_TXT: \u2935
   > # don't change anything below this point #
   > arm_64bit=1
 > enable_uart=1
   > uart_2ndstage=1
 > enable_gic=1
   > armstub=RPI_EFI.fd
 > disable_commandline_tags=1
   > disable_overscan=1
 > device_tree_address=0x1f0000
   > device_tree_end=0x200000
   > dtoverlay=miniuart-bt
CONFIG_TXT: \u2934

Not downloading /home/pi/wor-flasher-files/peinstaller - folder exists
Not downloading /home/pi/wor-flasher-files/driverpackage - folder exists
Not downloading /home/pi/wor-flasher-files/uefipackage - folder exists
Your system does not have enough usable disk space to generate a Windows 
image.
Please free up space or set the DL_DIR variable to a drive with more 
capacity.
11.8GB is necessary.
-- 

Regards and Thank You

John Williams
UK Mobile +447748334000
SKYPE  ejwffp

This e-mail has been scanned for viruses prior to transmission
The information contained in this e-mail is confidential and may also be 
subject to legal privilege. It is intended only for the recipient or 
recipients named above. If you are not named above as a recipient, you 
must not read, copy, disclose, forward or otherwise use the information 
contained in this email. If you have received this e-mail in error, 
please notify the sender immediately by reply e-mail and delete the 
message and any attachments without retaining any copies.

Botspot, 2021-09-23:
Not only does your destination drive need to be a certain size, but you also need a certain amount of free space to download files to.

ejwjohn, 2021-09-24:
OK, Thanks that was the key info.
Now I am running Windows 11 on the RPi4..thank You

Could I suggest that for other users like myself, it may be a good idea to add a little more info in the error message into what exactly the problem really is? . . . I thought I had read all the relevant documentation and  I could not find any other reference in the documentation in relation to minimum space required in order to actually download and create the image....

Thank You Again.....

JohnW

Botspot, 2021-09-24:
> OK, Thanks that was the key info.
> Now I am running Windows 11 on the RPi4..thank You
>
> Could I suggest that for other users like myself, it may be a good idea to add a little more info in the error [...]

I had intended for the error to be self-explanatory when it said Your **system** does not have enough **usable disk space** to **generate** a Windows image.

Would it be clearer if I changed the error to:
```
Your system needs 11.8GB of free space to download the Windows components to.
If your sd card is too small to do this, you can set the DL_DIR variable to a mounted drive with sufficient space. (Must be an Ext4 partition)
```

ejwjohn, 2021-09-24:
Hello,

Yes i believe that would be more informative, and for me, if I had read that I would have had clear direction on how to resolve it...

But as I said I am not an expert so others may believe your initial message is fine.

Thanks

JohnW

Botspot, 2021-09-24:
> But as I said I am not an expert so others may believe your initial message is fine.

Well WoR-flasher is supposed to be easy-to-use without any prior knowledge. I'll go ahead and improve the error. Thank you for reaching out and I hope you enjoy Windows 11.

closed 2021-09-24 by Botspot