Botspot/wor-flasher #44: no boot

NetModz, 2021-09-24:
Hello

i created a installation usb sandisk 32gb in latest raspbian in my Raspberry pi 4
everything finished well,i shutdown the system to start the installation of windows 11
but the screen does not turn on,speaks that have not a card in Rpi

any suggestion?

thanks

cycool29, 2021-09-24:
> i created a installation usb sandisk 32gb in latest raspbian in my Raspberry pi 4

Do you mean you installed Windows 11 in a USB?
If yes, you should enable USB boot in raspi-config before booting WIndows 11.

1. Boot your Raspbian SD card.
2. Open terminal, enter `sudo rpi-eeprom-update -asudo rpi-eeprom-update -a` to update the bootloader.
3. Next, type `sudo raspi-config` and enter,
4. Go to `Advanced options`, `Boot order`.
5. Choose `SD boot` (It will check if the SD card is inserted, if not, then will boot with USB, so do not stay your Raspbian SD plugged when booting Win11. If you need to do so, choose `USB Boot`.)
6. Shutdown your pi, unplug Raspbian SD card, insert your Win11 USB and power on.

I haven't tested with it but should works.
Hope it helps.

NetModz, 2021-09-24:
worflasher spoiled my raspberry pi 8gb

NetModz, 2021-09-24:
nothing appears on the hdmi and does not boot

NetModz, 2021-09-24:
broke the whole raspberry pi

Botspot, 2021-09-24:
> broke the whole raspberry pi

Just because it didn't boot a drive, doesn't mean it won't boot any drive. Try booting Raspberry Pi OS or something.

NetModz, 2021-09-24:
I switched to raspberry pi os doesn't work doesn't boot nothing appears on the monitor

NetModz, 2021-09-24:
turn on raspberry but nothing appears on the screen and the system does not boot

Botspot, 2021-09-24:
> I switched to raspberry pi os doesn't work doesn't boot nothing appears on the monitor

Does the Pi's green light flash at all? Does the Pi's red light flash at all?

NetModz, 2021-09-24:
my hdmi cable had broken

Botspot, 2021-09-24:
> my hdmi cable had broken

Is Windows successfully running on your Pi? Can this issue be closed?

NetModz, 2021-09-25:
i got it thanks for helping me

NetModz, 2021-09-25:
problem solved

closed 2021-09-25 by NetModz