Botspot/wor-flasher #45: Download Fail 503

ttrueten, 2021-09-25:
Trying to install Win10 (de-de) on RPI 4 but Download fails wiht Error 503

`9b2520|ERR |   508KiB/s|UUPs/microsoft-windows-client-desktop-required-arm64arm-package.esd
e19aaa|ERR |       0B/s|UUPs/professional_de-de.esd
d72cbe|ERR |   0.9MiB/s|UUPs/microsoft-windows-editionpack-professional-package.esd
13e9e6|ERR |       0B/s|UUPs/microsoft-windows-client-desktop-required-package.esd
8a06cb|ERR |       0B/s|UUPs/windows10.0-kb5005565-arm64.cab
906965|ERR |   0.9MiB/s|UUPs/microsoft.modernapps.clientn.all.esd
555136|ERR |   454KiB/s|UUPs/microsoft.modernapps.client.all.esd
ddd46f|ERR |   460KiB/s|UUPs/microsoft-windows-client-features-package.esd`

Botspot, 2021-09-25:
This looks like an issue with [uupdump](https://uupdump.net/).

Try using the en-us language. Once the windows setup starts you can change to your desired language.

ttrueten, 2021-09-25:
Thanks for your suggestion, but that doesn't solve the problem. I have tried various language settings, unfortunately all that does not work as well as an attempt to load Windows 11. Here's Linux Mint 20.1 running...:

`Download Results:
gid   |stat|avg speed  |path/URI
======+====+===========+=======================================================
ec9a2f|ERR |       0B/s|UUPs/microsoft-windows-not-supported-on-ltsb-arm64arm-package.esd
75c2d3|ERR |       0B/s|UUPs/microsoft-windows-foundation-package.esd
9ce448|ERR |       0B/s|UUPs/microsoft-windows-not-supported-on-ltsb-package.esd
6ba5c0|ERR |       0B/s|UUPs/microsoft-windows-languagefeatures-ocr-de-de-package-arm64.cab
df3090|ERR |       0B/s|UUPs/microsoft-windows-notepad-system-fod-package-arm64.cab
91c72c|ERR |       0B/s|UUPs/microsoft-windows-stepsrecorder-package-arm64.cab
20fdb9|ERR |       0B/s|UUPs/microsoft-windows-internetexplorer-optional-package-arm64.cab
72c4cf|ERR |       0B/s|UUPs/microsoft-windows-printing-pmcppc-fod-package-arm64.cab
69f829|ERR |       0B/s|UUPs/microsoft-windows-quickassist-package-arm64.cab
e490af|ERR |       0B/s|UUPs/microsoft-windows-wordpad-fod-package-arm64.cab
616fc4|ERR |       0B/s|UUPs/microsoft-windows-powershell-ise-fod-package-arm64.cab
0a13f1|ERR |       0B/s|UUPs/openssh-client-package-arm64.cab
6c24e5|ERR |       0B/s|UUPs/microsoft-windows-required-shellexperiences-desktop-wow64-package.esd
2dc03b|ERR |       0B/s|UUPs/microsoft-windows-required-shellexperiences-desktop-arm64arm-package.esd
a92a40|ERR |       0B/s|UUPs/windows10.0-kb5005662-arm64.cab
870070|ERR |       0B/s|UUPs/microsoft-onecore-applicationmodel-sync-desktop-fod-package-arm64.cab
8d1ec0|ERR |       0B/s|UUPs/microsoft-windows-editionspecific-professional-package.esd
087815|ERR |       0B/s|UUPs/microsoft-windows-languagefeatures-handwriting-de-de-package-arm64.cab
ff358b|ERR |       0B/s|UUPs/microsoft-windows-tabletpcmath-package-arm64.cab
28f3e7|ERR |       0B/s|UUPs/microsoft-windows-required-shellexperiences-desktop-package.esd
968ea4|ERR |       0B/s|UUPs/microsoft-windows-editionpack-professional-arm64arm-package.esd
435f97|ERR |       0B/s|UUPs/microsoft-windows-editionpack-professional-wow64-package.esd
f15bb8|ERR |       0B/s|UUPs/microsoft-windows-wowpack-corearm-arm64arm-package.esd
f4f491|ERR |       0B/s|UUPs/ssu-22000.190-arm64.cab
1e518e|ERR |       0B/s|UUPs/microsoft-windows-mediaplayer-package-arm64.cab
01cc5d|ERR |       0B/s|UUPs/microsoft-windows-regulatedpackages-arm64arm-package.esd
329c65|ERR |       0B/s|UUPs/microsoft-windows-client-languagepack-package_de-de-arm64-de-de.esd
19e53b|ERR |       0B/s|UUPs/microsoft-windows-regulatedpackages-wow64-package.esd
bc05dc|ERR |       0B/s|UUPs/microsoft-windows-userexperience-desktop-package-arm64.cab
36e5c4|ERR |       0B/s|UUPs/microsoft-windows-not-supported-on-ltsb-wow64-package.esd
365105|ERR |       0B/s|UUPs/microsoft-windows-languagefeatures-basic-de-de-package-arm64.cab
d76973|ERR |       0B/s|UUPs/microsoft-windows-regulatedpackages-package.esd
de6608|ERR |       0B/s|UUPs/microsoft-windows-languagefeatures-texttospeech-de-de-package-arm64.cab
7f4b16|ERR |       0B/s|UUPs/microsoft.modernapps.client.professional.esd
7251e1|ERR |       0B/s|UUPs/microsoft-windows-chpe-binaries-arm64x86-package.esd
49169d|ERR |       0B/s|UUPs/windows10.0-kb5005635-arm64.cab
a4336f|ERR |       0B/s|UUPs/microsoft-windows-languagefeatures-speech-de-de-package-arm64.cab
bb8f34|ERR |       0B/s|UUPs/microsoft-windows-hello-face-package-arm64.cab
1572b3|ERR |       0B/s|UUPs/microsoft-windows-client-features-arm64arm-package.esd
42d546|ERR |       0B/s|UUPs/microsoft-windows-client-features-wow64-package.esd
eeee54|ERR |       0B/s|UUPs/microsoft-windows-editionpack-professional-package.esd
41790f|ERR |       0B/s|UUPs/windows10.0-kb5005635-arm64.psf
189dc4|ERR |       0B/s|UUPs/microsoft-windows-client-desktop-required-arm64arm-package.esd
bbde1e|ERR |       0B/s|UUPs/microsoft-windows-client-desktop-required-wow64-package.esd
722496|ERR |       0B/s|UUPs/microsoft-windows-client-features-package.esd
69e266|ERR |       0B/s|UUPs/microsoft.modernapps.client.all.esd
10128e|ERR |       0B/s|UUPs/professional_de-de.esd
08c239|ERR |       0B/s|UUPs/microsoft-windows-client-desktop-required-package.esd

Status Legend:
(ERR):error occurred.
`

cycool29, 2021-10-04:
Hmm... from the output you've given, the aria2c don't even download the packages.
Do you have any firewall or something that might block the downloads?
Try `wget 'https://uupdump.net/` to test if [uupdump](https://uupdump.net/) is blocked.

ttrueten, 2021-10-04:
Thanks for th hint, that solved it. [SNORT](https://www.snort.org) apparently blocked some of the download links. The download and baking of the ISO file worked. Thank you.
[image]