Botspot/wor-flasher #47: It says i dont have enough storage

skuxkid123, 2021-10-06:


cycool29, 2021-10-06:
`DL_DIR` variable is the directory you want to download the installation components before flashing you SD card, and it defaults to ` ~/wor-flasher-files`.

This error occurs as you don't have sufficient space to download the installation components in your system drive.
The solution is plug another storage and change the `DL_DIR` variable to the storage to store the files. 

To change `DL_DIR` variable:

## `GUI`

At the `Installation Overview` page, click `Advanced`. 

[image]

Now you can change `DL_DIR` referred as `Working directory` to the storage you want to store the installaton components.

[image]

## `CLI`

Before running `~/wor-flasher/install-wor.sh`, enter `DL_DIR=/path/to/your/storage`, and the script will obey that variable.

cycool29, 2021-10-18:
@skuxkid123 Any updates?

skuxkid123, 2021-12-23:
Yeah it worked a couple of times but then I had to turn it off but now I get this error

An error has occurred:
list_langs(): requires an argument for windows update ID. Example ID: db8ec987-d136-4421-afb8-2ef109396b00
Exiting now.

please reply asap on how to fix

Cpmjaguar1234, 2022-01-30:
Ur mom

Botspot, 2022-07-01:
> Yeah it worked a couple of times but then I had to turn it off but now I get this error
> 
> An error has occurred: list_langs(): requires an argument for windows update ID. Example ID: db8ec987-d136-4421-afb8-2ef109396b00 Exiting now.
> 
> please reply asap on how to fix

That would be a site outage, and I have since made multiple improvements to reduce the chance of that error and to better explain the error.
Closing this issue.