Botspot/wor-flasher #52: Failed to determine a hashsum error

Dialtone65, 2021-10-19:
The graphics interface comes up when trying to install Windows but I get an error:
An error has occured:
Failed to determine a hashsum for WoR PE-based installer.
URL:http://worproject.ml/dldserv/worpe/gethslatest.php

Anyone have a suggestion on how to get past this?
Thanks,

luxadi, 2021-11-11:
I'm also getting this error. I think the worproject.ml server might be down at the moment. Did it end up working for you @Dialtone65 ?

Dialtone65, 2021-11-11:
I found an article on how to update the server list and after selecting a new server, it worked.

luxadi, 2021-11-11:
Oh that's great, do you remember which article?

Dialtone65, 2021-11-11:
Here is a link to the web page I followed:
https://www.windowslatest.com/2020/07/24/install-windows-10-on-raspberry-pi/
Here is the link to download WoR:
https://www.worproject.ml/downloads
Basically I installed WoR on my windows machine, and using the Windows 10 image, I flashed it to a 32Gb memory card and then did some minor configuration from there.

Botspot, 2022-02-10:
With commit https://github.com/Botspot/wor-flasher/commit/eabdb334ab978a673588c1e00facead0d75f214c, the `wget` command tries harder to download files and has a higher chance of succeeding.

I will close this issue now, assuming this change fixed the problem.

closed 2022-02-10 by Botspot