Botspot/wor-flasher #56: [error] File: B:\Windows\WinSxS\arm64_dual_wdmaudio.inf_31bf3856ad364e35_10.0.19041.1_none_64bcd5b0eb144ad9\drmkaud.sys

cgioconda, 2021-11-04:
Receiving the following error with both 

Feature update to Windows 10, version 21H1 (19043.1288)| 01046126-ae3f-4711-a339-4ed8e08af2cc
and
Cumulative Update Preview for Windows 10 Version 21H1 (19043.1320)| e01fae77-a080-402c-b7a3-686cc990c115

2021-10-19 19:05:58.6911 [debug] Exit code: 0
2021-10-19 19:05:58.7251 [info] Windows partition created! (letter: B)
2021-10-19 19:05:58.7489 [info] Applying the WIM image to: B
2021-10-19 19:25:24.6645 [error] File: B:\Windows\WinSxS\arm64_dual_wdmaudio.inf_31bf3856ad364e35_10.0.19041.1_none_64bcd5b0eb144ad9\drmkaud.sys (433)
2021-10-19 19:25:30.8672 [error] File: B:\Windows\WinSxS\arm64_microsoft-windows-f..e-microsoftjhenghei_31bf3856ad364e35_10.0.19041.1_none_1b31ce407f726d4a\msjh.ttc (21)
2021-10-19 19:25:31.4940 [error] System.ComponentModel.Win32Exception (433): A device which does not exist was specified.
   at Microsoft.Wim.WimgApi.ApplyImage(WimHandle imageHandle, String path, WimApplyImageOptions options) in Microsoft.Wim.dll:token 0x6000020+0x3a
   at WoR_PE.Wrappers.WIMGAPIWrapper.<>c__DisplayClass6_0.<ApplyImage>b__0() in WoR-PE.dll:token 0x60001c1+0x52
   at System.Threading.Tasks.Task.InnerInvoke() in System.Private.CoreLib.dll:token 0x6002d62+0xf
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state) in System.Private.CoreLib.dll:token 0x60029bb+0x4c
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread) in System.Private.CoreLib.dll:token 0x6002d61+0x12e
--- End of stack trace from previous location ---
   at WoR_PE.Wrappers.WIMGAPIWrapper.ApplyImage(String imageFile, String applyPath, UInt32 imageIndex, ProgressCallback callback) in WoR-PE.dll:token 0x6000084+0x9a
   at WoR_PE.Setup.SetupProcess.ApplyImageTask() in WoR-PE.dll:token 0x60000e3+0xe8
   at WoR_PE.Setup.SetupProcess.DeploymentTask() in WoR-PE.dll:token 0x60000de+0x29b
   at WoR_PE.Setup.SetupProcess.Run(IWin32Window parent) in WoR-PE.dll:token 0x60000dc+0xa7
2021-10-19 19:25:31.6781 [warn] The installation could not be finished!
2021-10-19 19:26:23.8984 [info] Elapsed time: 00:21:35

cgioconda, 2021-11-05:
Used a different, older release and it worked.

closed 2021-11-05 by cgioconda

Botspot, 2021-11-05:
> Used a different, older release and it worked.

Yes, sometimes uupdump doesn't work as a new release rolls in. Using an older update ID usually fixes it.