Botspot/wor-flasher #57: Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST

B2025, 2021-11-06:
## Setup:

- Which version of Windows you chose
- I chose windows 11
- Raspberry Pi model
- Pi 4
- Name of SD card you tried to flash (some are too slow or too small)
- samsung 860 evo
- What day you first downloaded wor-flasher
- 11/05/2021
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
- Raspberry pi os

## Procedure:
 when I tried To run WoR-flasher using the graphical interface it ran and at the end it said the following:

Retrieving aria2 script...

11/06 00:05:49 [NOTICE] Downloading 1 item(s)

11/06 00:05:49 [WARN] aria2c had to connect to the other side using an unknown TLS protocol. The integrity and confidentiality of the connection might be compromised.
Peer: uupdump.net (104.21.88.160:443)

11/06 00:05:50 [NOTICE] Download complete: /home/pi/wor-flasher-files/uupdump/aria2_script.4638.txt

Download Results:
gid   |stat|avg speed  |path/URI
======+====+===========+=======================================================
ecb4c3|OK  |      81B/s|/home/pi/wor-flasher-files/uupdump/aria2_script.4638.txt

Status Legend:
(OK):download completed.
Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST
If this problem persists, most likely the set you are attempting to download was removed from Windows Update servers.

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 4 of 4)
Failed to generate a Windows ISO! uup_download_linux.sh exited with an error so please see the errors above.

cycool29, 2021-11-06:
There is a [similar issue](https://github.com/uup-dump/website/issues/44).

RUHavingFun, 2021-11-11:
> There is a [similar issue](https://github.com/uup-dump/website/issues/44).

I am not sure of the fix based on that link.  Also, I had already run this successfully on a SSD drive and was running again to put on another drive.  I assume I already have the files needed on the SD card in the wor-flasher-files folder.  Trying to understand what I need to do to make this run again.

jadevoyd, 2021-11-25:
I am using the same configuration as B2025 with the same failed result.  I have tried an M.2 drive in an external case as well as brand new Micro SD cards.  It never gets to the flashing process, it stops at the download.
I have tried the fix cycool29 mentioned, but I get the same result - the config change does not stick and gets automatically reset somewhere in the process

Botspot, 2022-02-10:
If this error still happens, let me know and I'll bump the Windows 11 update ID.

B2025, 2022-02-11:
It’s fixed

abdulqadeerqureshi, 2023-04-14:
Getting this error for Windows 10 21H2 UUP dump.

```
Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST
If this problem persists, most likely the set you are attempting to download was removed from Windows Update servers.

```

ksingh7, 2023-04-26:
Getting this error with [UUP dump (Windows 10 21H2)](https://uupdump.net/known.php?q=21390)
```
(OK):download completed.
Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST
If this problem persists, most likely the set you are attempting to download was removed from Windows Update servers.
```

Botspot, 2023-04-26:
Same for you, @abdulqadeerqureshi.

Josesosa07, 2023-04-27:
I got same issue, how can I update?
@Botspot

Botspot, 2023-04-27:
> I got same issue, how can I update? @Botspot

@Josesosa07 WoR-flasher is supposed to auto-update every time you run it.
To help me understand what is going wrong, please run these commands and report the output back to me.
```
cd ~/wor-flasher
git rev-parse HEAD
git ls-remote https://github.com/Botspot/wor-flasher HEAD | awk '{print $1}'
```

Botspot, 2023-04-29:
Due to @Josesosa07 not replying, would @ksingh7 or @abdulqadeerqureshi be willing to run the commands I asked @Josesosa07 to run?

punz1738, 2023-05-07:
running the last command I get 'ad85b75b7552408c2a475f02b24ff0f3a3af697a'
 
im having this issue using windows 10 21h2. the other two commands don't run or it might be I don't know how to run them

Botspot, 2023-05-10:
> running the last command I get 'ad85b75b7552408c2a475f02b24ff0f3a3af697a'
> 
> im having this issue using windows 10 21h2. the other two commands don't run or it might be I don't know how to run them

@punz1738, did running the `git rev-parse HEAD` command not return anything?
Please also run this command: `ls -A ~/wor-flasher` and reply here on github what that command said.

punz1738, 2023-05-10:
Returns nothing. I am running this on the Mac terminal which might be the reason Im not returning anything. OP is using this for the raspberry pi but I was trying to convert windows 10 21h2 from UUP dumb into an iso for parallels.

Botspot, 2023-05-10:
> Returns nothing. I am running this on the Mac terminal which might be the reason its not returning anything. OP is using this for the raspberry pi but I was trying to convert windows 10 21h2 from UUP dump into an iso for parallels.

Mac Os does some things differently than linux. I don't have a Mac OS device to test on, so your mileage may vary. Furthermore, the fact that `git rev-parse HEAD` returns nothing means that you did not use `git clone` to download wor0flasher. You may instead have downloaded a ZIP release and extracted it. Or maybe you deleted the `.git` folder. No matter what happened, you managed to break wor-flasher's update mechanism.

You mentioned that you want to use wor-flasher to generate an ARM64 ISO for you using uupdump. The latest wor-flasher does not use uupdump anymore, nor does it generate an ISO, so it will not be very helpful for you in this case. I had to rewrite wor-flasher to ***not*** use uupdump when they took their site down. Uupdump is back online now, but their API is still offline. WoR-Flasher now uses Windows ESD releases instead of uupdump.

@punz1738, if you want a Windows ISO from uupdump, I suggest you use uupdump directly. It's not that hard. See https://uupdump.net

Botspot, 2023-05-11:
Just wanted to remind everyone encountering this issue that if you see the "EMPTY_FILELIST" error, it means your version of wor-flasher is out of date.

happyguo4614590, 2023-07-03:
I am having the same problem.

Status Legend:
(OK):download completed.
Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST
If this problem persists, most likely the set you are attempting to download was removed from Windows Update servers.

Botspot, 2023-07-03:
> I am having the same problem.
> 
> Status Legend: (OK):download completed. Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST If this problem persists, most likely the set you are attempting to download was removed from Windows Update servers.

@happyguo4614590, please run this command to see how outdated your version of wor-flasher is.
```
cd ~/wor-flasher; git show -s --format="%ad" --date=short | xargs date +%x -d
```

3Percs4Zoid, 2023-07-06:
> git ls-remote https://github.com/Botspot/wor-flasher HEAD | awk '{print $1}'

this one did nothing and the others got errors

Botspot, 2023-07-06:
> > git ls-remote https://github.com/Botspot/wor-flasher HEAD | awk '{print $1}'
> 
> this one did nothing and the others got errors

@3Percs4Zoid, please copy the errors here.

charles9945, 2023-11-14:
Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST
If this problem persists, most likely the set you are attempting to download was removed from Windows Update servers.

I get this when running on m2 MacBook Air any help

jredfox, 2024-01-20:
I am running from `Version 10.0.19045.3930` host and I get the same exact error. I tried from the website today

closed 2024-02-10 by Botspot