Botspot/wor-flasher #59: Issues downloading files.

ve3elb, 2021-11-27:
Hello everyone I am trying to Wor-Flasher but having issues when it get to uup_download_linux.sh. Internet connection is 100% fine only getting this issue when I get to this step. Any ideas please help. Thank you. 

error:
uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 4 of 4)
Failed to generate a Windows ISO! uup_download_linux.sh exited with an error so please see the errors above.

EddyBeaupre, 2021-11-27:
Same issue:

Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST
If this problem persists, most likely the set you are attempting to download was removed from Windows Update servers.

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 4 of 4)
Failed to generate a Windows ISO! uup_download_linux.sh exited with an error so please see the errors above.

[aria2_download.log](https://github.com/Botspot/wor-flasher/files/7612272/aria2_download.log)

Fishin13, 2021-11-28:
I am having the same issue

Generating Windows image with uupdump
Retrieving aria2 script...

11/28 14:58:04 [NOTICE] Downloading 1 item(s)
[#5f3483 0B/0B CN:1 DL:0B]                                                     
11/28 14:58:06 [NOTICE] Download complete: /home/pi/wor-flasher-files/uupdump/aria2_script.27520.txt

Download Results:
gid   |stat|avg speed  |path/URI
======+====+===========+=======================================================
5f3483|OK  |      23B/s|/home/pi/wor-flasher-files/uupdump/aria2_script.27520.txt

Status Legend:
(OK):download completed.
Unable to retrieve data from Windows Update servers. Reason: EMPTY_FILELIST
If this problem persists, most likely the set you are attempting to download was removed from Windows Update servers.

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 1 of 4)

Botspot, 2021-11-29:
Thanks for reporting. I've been able to reproduce the error. Basically, the script has used a hard-coded Windows update ID because the uupdump website [malfunctioned](https://github.com/Botspot/wor-flasher/issues/41) on fresher releases. Well it looks like now that older version is no longer available.

I tried using the very latest update ID (`65b6f583-d9c4-4ef2-8259-7ca0d63c263b`), but that failed due to a missing language list. So I've bumped the hard-coded version to `fdb8a3c7-e28d-48e2-80fa-54673f5786ee` with commit https://github.com/Botspot/wor-flasher/commit/10ee37b7f723fcf4bccd5280b675cf26ff613668.

For now, the problem is solved, but eventually this update ID will also go offline. So we've basically kicked the can down the road. ;) Hopefully by then the [uupdump website](https://uupdump.net/) will have its issues sorted out.

closed 2021-11-29 by Botspot