Botspot/wor-flasher #61: No languages in that selection stage

skuxkid123, 2021-12-23:
## Setup:

- Which version of Windows you chose 10 similar for 11 
- Raspberry Pi model 4b 2gb
- Name of SD card you tried to flash (some are too slow or too small) sandisk Cruzer glide 32gb
- What day you first downloaded wor-flasher idk
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc) Raspberry Pi OS

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup. I tried to install via wor flasher on pi apps but exact same thing happened in the terminal version I selected windows 10 and pi4/400, but when I clicked next it came up with the error in a little window 

An error has occurred:
list_langs(): requires an argument for windows update ID. Example ID: db8ec987-d136-4421-afb8-2ef109396b00
Exiting now.

But when I hit okay it comes up with the language selection stage but only has two languages one saying FALSE and the other is blank but has check box. Even when I tried switching to installing windows 11, it went straight to the broken language page that said the FALSE option and the blank option. Please respond asap.

Botspot, 2021-12-24:
The list of languages is fetched from [the UUPdump website](https://uupdump.net). If you're not connected to the Internet, the list of languages will fail to display properly.
Sometimes UUPdump goes offline for maintenance, but I just launched WoR-Flasher myself and the language selection window was working properly. Please be sure you are connected to the Internet and that the website isn't blocked.

skuxkid123, 2021-12-25:
Nevermind i Switches OSes and works fine now

closed 2021-12-27 by Botspot