Botspot/wor-flasher #63: Shitty Error start PXE over ipv4

Atten007, 2021-12-25:
Hi folks,

I‘m currently trying to install Windows 11 on my Raspberry Pi 4B, BUT when I‘m pressing enter in the menu with the raspberry image, the very shitty error with „Start PXE over IPV4“ and „PXE-E16: A valid offer received“ occurs. If you can tell me, how I can fix this very shitty error, it’d be really nice for me!!! And sorry for the word shitty, I’m writing it here because it really annoys me that my Windows 11 installation doesn’t work.

Sincerely yours,
Atten007😊

Atten007, 2021-12-25:
[image]

Botspot, 2021-12-27:
By the way, your image failed to upload and I was unable to see it.

Usually this error indicates a slow SD card and/or a failed flash. I suggest you go through the flashing process again to see if it helps.
See [this issue](https://github.com/Botspot/wor-flasher/issues/30).

Risae, 2022-01-02:
I also have this issue. After the WOR Installation to the drive was completed, the Raspberry Pi restarted itself and said that it wants to boot over PXE. 2 times it tried PXE over TCP or something, then it changed to PXE over HTTP and fronze.
After unplugging the ethernet cable and restarting the Raspberry Pi it seemed to properly boot into Windows.
Maybe Raspi UEFI changed the boot order to PXE?

ajorg, 2022-02-08:
I believe I ran into the same error. After waiting for PXE and other options to fail, I rebooted and pressed ESC to go to setup, then selected the SD / MMC drive, skipping PXE, and it booted. It seemed to me that the problem may be simply that other options are tried first, and then it takes a long time to boot. I couldn't see how to configure it to boot from SD first.

Mine was installed using an older version, some time ago, so I don't know if this might be fixed lately.

folgers, 2022-06-22:
I kept getting the PXE boot error when I was trying to use an SD card.  Once I used a USB SATA for boot, I never saw that error again.  I am now happily using Windows 11 via USB SATA boot and will never go back to an SD card.

Botspot, 2022-07-01:
Closing this issue due to inactivity.

closed 2022-07-01 by Botspot

vaisd007302, 2024-12-03:
I still have the same problem.