Botspot/wor-flasher #65: Error installing windows on SD

seanmcilvenna, 2022-01-02:
## Setup:

- Which version of Windows you chose: 22000.1_PROFESSIONAL_ARM64_EN-US
- Raspberry Pi model: 4
- Name of SD card you tried to flash (some are too slow or too small): Samsung 32GB EVO Select
- What day you first downloaded wor-flasher: 1/2/2022
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc): Raspberry Pi POS

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

1. From Raspberry Pi OS, cloned repo
2. Inserted USB drive to use as working drive because Raspberry Pi OS disk is too small to download files
3. Ran `install-wor-gui.sh`
4. Selected Language, OS, etc.
5. Selected "Advanced" and chose working directory on USB drive
6. When finished flashing, shutdown Pi and swaped MicroSD with the newly flashed one
7. Waited forever... Eventually stalled at 21%. Looked at logs, found these errors, and 30 minutes later it didn't progress pass the errors.

[WoR-PE.log](https://github.com/Botspot/wor-flasher/files/7799784/WoR-PE.log)

seanmcilvenna, 2022-01-02:
```
Leaving DiskPart...

2021-10-19 19:22:36.6168 [debug] Exit code: 0
2021-10-19 19:22:36.6494 [info] Windows partition created! (letter: B)
2021-10-19 19:22:36.6634 [info] Applying the WIM image to: B
2021-10-19 19:33:26.6590 [error] File: B:\Windows\WinSxS\arm64_microsoft-windows-d..ers-local.resources_31bf3856ad364e35_10.0.22000.1_en-us_a209e35c58b17508\FfuProvider.dll.mui (121)
2021-10-19 19:35:35.5946 [error] File: B:\Windows\WinSxS\arm64_microsoft-windows-g..linetools.resources_31bf3856ad364e35_10.0.22000.1_en-us_ea567c87ab2b785e\gpupdate.exe.mui (121)
```

seanmcilvenna, 2022-01-02:
I also tried Windows 10... and getting a similar error, but for different files
[WoR-PE.log](https://github.com/Botspot/wor-flasher/files/7799913/WoR-PE.log)
.

seanmcilvenna, 2022-01-02:
Waited another 30-45 minutes and eventually the windows 10 installation failed with the following logs...
[WoR-PE.log](https://github.com/Botspot/wor-flasher/files/7799970/WoR-PE.log)

Botspot, 2022-01-03:
I'm afraid I'm unable to help much. The Win-PE environment is mostly Microsoft's code, and the rest was written by WoR developers. Anything that fails during that step is usually a mystery.
From past experience though, we have noticed that many Samsung EVO cards do fail in the Win-PE environment. Faster cards like SanDisk Extreme perform much better.

So my advice would be to get a faster SD card.

closed 2022-01-03 by Botspot

seanmcilvenna, 2022-01-03:
So, can anyone confirm that this wor-flasher works with the latest builds on https://uupdump.net/?