Botspot/wor-flasher #66: About errors in Ubuntu 18

ncs19960831, 2022-01-03:
In the ubuntu 18 LTS, I encountered a disk read error.
After checking, I found that the "PATH" parameter could not be recognized in lsblk, like "lsblk -I 8,179,259 -dno PATH".
[image]

lsblk would throw an "unknown item" error . And I tried the Ubuntu21 version, this problem does not occur.
[image]

lsblk is 2.31.1 in ubuntu18, and 21 is 2.36.1. I don’t know which version can parse the PATH parameter. 
[image]
[image]

I tried it. You can use NAME to replace the PATH parameter in ubuntu 18, or throw a detailed error 
[image]

Cpmjaguar1234, 2022-01-30:
UwU

Botspot, 2022-02-10:
With commit https://github.com/Botspot/wor-flasher/commit/eabdb334ab978a673588c1e00facead0d75f214c#diff-d99d6dcec506a8352af9322c4e6a1db3622e63a31f2b1b0ba27731130f381b8bR144, I've made an initial fix to this problem.

@ncs19960831, could you try running WoR-Flasher again to see if it works now?

commit 6c74b0ce (2022-07-01): get_partition: use lsblk NAME option to close #66