Botspot/wor-flasher #69: Legal

Maki2007, 2022-01-28:
Is this legal at all? 
This is not something there hacks into ms server, to get the OS?
I dont want to get a big thicket.

Botspot, 2022-01-28:
Short answer: Yes.

Microsoft doesn't provide a ready-made ARM image (yet), but it does publicly distribute all of the components necessary to build one. Microsoft mainly does this so that large computer/laptop manufacturers can build their own customized Windows images, but it's perfect for WoR-Flasher as well.

If you use this flasher, Windows will ask you for a license key, just like any other ordinary installation.
Of course, I may be **lying**. How can you trust me? In that case, go read through [this discussion](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=318599) on the Raspberry Pi Forums. Several folks were arguing that this was illegal, but eventually some Raspberry Pi employees came out and [confirmed](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=318599#p1907313) that WoR-Flasher is perfectly legal.

closed 2022-01-28 by Botspot

Maki2007, 2022-01-31:
Ok, now I understand how it is

Botspot, 2022-01-31:
@Maki2007, Thank you for reaching out. If you need anything else, please let me know!

Maki2007, 2022-02-01:
Ok thanks for help:-)