Botspot/wor-flasher #72: How does it work? (Not a issue, but a question)

Maki2007, 2022-04-18:
I had tried runnning win11 arm on qemu on bullseye host (not succes). 
The BIOS/EFI(call it whatever you want) using this is the same as using qemu on a RPI and virtualise it. 
Is this “just”(full respect for the work) a fully windowed qemu virtualisation and not running native on the pi?
If it is, what OS is the host? #

Botspot, 2022-04-18:
> I had tried runnning win11 arm on qemu on bullseye host (not succes).

If you ever do succeed with that, let me know! I'd like to app-ify that for those who don't have multiple pies.

> Is this “just”(full respect for the work) a fully windowed qemu virtualisation and not running native on the pi? If it is, what OS is the host? #

Not quite. UEFI is an OS compatibility layer that provides a standardized interface, no matter the underlying hardware. Windows 10/11 is UEFI-compatible for the ARM architecture.
Upon power-up, UEFI starts first. (This is the black screen with the large RPi logo) When finished, UEFI will start Windows, which runs on top of UEFI.

I wouldn't be too surprised if UEFI is using some baremetal Linux internally (after all, the Pi4's GPU bootloader runs a weird, closed-source linux distro as well), but I do know that this is more than a clever usage of QEMU.

Maki2007, 2022-04-18:
Okay thanks for quick answer
I will let you know if I ever get qemu virtualisation working. Will you upload it to pi-apps, if you can “app-ify”?
Are you on full time with raspberry pi, because you always answer really quick and make some very high quality programs for the raspberry pi?

Botspot, 2022-04-18:
> Will you upload it to pi-apps, if you can “app-ify”?

It would be similar format to WoR-Flasher - having its own github repo. But yes, I would also add it to Pi-Apps.

> Are you on full time with raspberry pi?

Yes. I used a Pi3 as my main PC, then upgraded to a 4GB Pi4 when it was released. I haven't touched a conventional PC in like 4 years.
Keep me updated on your QEMU endeavors. ;)

closed 2022-04-18 by Botspot

Maki2007, 2022-04-19:
I have a problem with QEMU. By some reason it won’t allocate ram to the VM. When  I am in UEFI settings there 0 mb of ram. I had tried just waiting but it didn’t work. 
But I will find a solution:-)

Botspot, 2022-04-19:
> I have a problem with QEMU. By some reason it won’t allocate ram to the VM. When I am in UEFI settings there 0 mb of ram. I had tried just waiting but it didn’t work. But I will find a solution:-)

Do you have Discord? If so, I recommend messaging `setLillie#5644`. SetLillie claims to have gotten Windows 10 working with QEMU, and even wrote instructions. Unfortunately I haven't been to reproduce setLillie's results, but perhaps they could help you.

I'm also aware of this thread on the RPi Forums. https://forums.raspberrypi.com/viewtopic.php?t=248345

Maki2007, 2022-04-19:
Thanks for sending the links
If I get it work, I can’t take the honour. 
I don’t have discord.

Botspot, 2022-04-19:
> I don’t have discord.

OK here's SetLillie's latest instructions:
Some steps appear to be missing. Good luck!

> On ARM64 device (and OS)
> ```bash
> #!/bin/bash
> mkdir -p ${HOME}/win10/
> cd ${HOME}/win10/
> qemu-img create -f qcow2 disk.img 64G
> wget https://gist.github.com/niw/4f1f9bb572f40d406866f23b3127919b/raw/f546faea68f4149c06cca88fa67ace07a3758268/QEMU_EFI-cb438b9-edk2-stable202011-with-extra-resolutions.tar.gz
> tar xzvf QEMU_EFI-*.tar.gz
> dd if=/dev/zero of=flash0.img bs=1M count=64
> dd if=/dev/zero of=flash1.img bs=1M count=64
> dd if=QEMU_EFI.fd of=flash0.img conv=notrunc
> dd if=QEMU_VARS.fd of=flash1.img conv=notrunc
> ```
> Download for drivers: 
> https://fedorapeople.org/groups/virt/virtio-win/direct-downloads/archive-virtio/virtio-win-0.1.215-2/virtio-win.iso
> 
> to start (raspberry pi):
> ```bash
> #!/bin/bash
> cd ${HOME}/win10/
> qemu-system-aarch64 \
> -cpu host \
> -enable-kvm \
> -M virt,accel=kvm \
> -smp 3 \
> -m 2G \
> -drive file=flash0.img,format=raw,if=pflash \
> -drive file=flash1.img,format=raw,if=pflash \
> -device VGA -global VGA.vgamem_mb=256 -display gtk,gl=on \
> -device qemu-xhci \
> -device usb-kbd \
> -device usb-mouse \
> -device usb-tablet \
> -device usb-audio \
> -device usb-storage,drive=install \
> -drive if=none,id=install,format=raw,media=cdrom,file=install.iso \
> -device usb-storage,drive=drivers \
> -drive if=none,id=drivers,format=raw,media=cdrom,file=drivers.iso \
> -device virtio-blk,drive=system \
> -drive if=none,id=system,format=qcow2,discard=unmap,cache=none,file=disk.img \
> -device virtio-net,netdev=net0 \
> -netdev user,id=net0,hostfwd=tcp::5555-:3389
> ```
> once booted, when it says press any key to continue: press a key and WAIT
> now that windows 10 is installed, you should edit the file to remove the 2 lines with the install.iso.
> 
> Now that, that's done: you can install some drivers. 
> 
> Open command prompt as administrator:
> ```
> bcdedit /set TESTSIGNING ON
> shutdown /r /t 0
> ```
> (The second command reboots, you can just use the buttons too)
> 
> now you're rebooted into test mode: go to the file manager (explorer.exe) and find your drivers cd. 
> NETKVM/w10/arm64/
> find the file that you'd usually use to install a driver, right click: install, now you have networking.
> ```
> bcdedit /set TESTSIGNING OFF
> shutdown /r /t 0
> ```
> now test signing is off and you're back in windows 10, have fun!
> If your UEFI doesn't display: 
> 
> UEFI not displaying fix:
> ```
> #!/bin/bash
> cd ${HOME}/win10/
> qemu-system-aarch64 \
> -cpu host \
> -enable-kvm \
> -M virt,accel=kvm \
> -smp 3 \
> -m 2G \
> -drive file=flash0.img,format=raw,if=pflash \
> -drive file=flash1.img,format=raw,if=pflash \
> -device ramfb \
> -device qemu-xhci \
> -device usb-kbd \
> -device usb-mouse \
> -device usb-tablet \
> -device usb-audio \
> -device usb-storage,drive=install \
> -drive if=none,id=install,format=raw,media=cdrom,file=install.iso \
> -device usb-storage,drive=drivers \
> -drive if=none,id=drivers,format=raw,media=cdrom,file=drivers.iso \
> -device virtio-blk,drive=system \
> -drive if=none,id=system,format=qcow2,discard=unmap,cache=none,file=disk.img \
> -device virtio-net,netdev=net0 \
> -netdev user,id=net0,hostfwd=tcp::5555-:3389
> ```

Botspot, 2022-04-19:
Looks like this is the build commands setlillie used for compiling qemu:

> ```bash
> sudo apt-get install git-email libaio-dev libbluetooth-dev libbrlapi-dev libbz2-dev libcap-dev libcap-ng-dev libcurl4-gnutls-dev libgtk-3-dev libibverbs-dev libjpeg62-turbo-dev libncurses5-dev libnuma-dev librbd-dev librdmacm-dev libsasl2-dev libsdl1.2-dev libseccomp-dev libsnappy-dev libssh2-1-dev libvde-dev libvdeplug-dev libxen-dev liblzo2-dev valgrind xfslibs-dev ninja-build
> sudo apt install -y build-essential libepoxy-dev libdrm-dev libgbm-dev libx11-dev libvirglrenderer-dev libpulse-dev libsdl2-dev
> wget https://download.qemu.org/qemu-6.2.0.tar.xz 
> tar xf qemu-6.2.0.tar.xz
> cd qemu-6.2.0/
> ./configure --enable-sdl  --enable-opengl --enable-virglrenderer --enable-system --enable-modules --audio-drv-list=pa --enable- kvm --enable-gtk --target-list=aarch64-softmmu
> ninja -C build
> sudo ninja -C build install 
> ```
> compile QEMU script... will only enable the aarch64 core btw

Maki2007, 2022-04-19:
How can some steps be missing?

Sendt fra mit Apple Watch

Maki2007, 2022-04-19:
The thread on RPi forum is too confusing

Sendt fra mit Apple Watch

Maki2007, 2022-04-20:
Right now I have a really annoying problem. 
You know the GUI with “Machine” and “View” in QEMU. It’s gone, and I don’t know the keyboard shortcut to get it back again:-(

Botspot, 2022-04-20:
> Right now I have a really annoying problem. You know the GUI with “Machine” and “View” in QEMU. It’s gone, and I don’t know the keyboard shortcut to get it back again:-(

It's possible that you're using a command-line flag to hide the menubar.

Maki2007, 2022-04-21:
Good news. I got the installer booting up. I only need these virtIO drivers

Maki2007, 2022-04-21:
Do you have any ideas how to get theese virtIO driver on the machine?

FlameKat53, 2022-04-22:
Hello. Running windows 10 or 11? 

here's the absolute latest Virtio drivers I see they now are with windows 11 arm support: https://fedorapeople.org/groups/virt/virtio-win/direct-downloads/archive-virtio/virtio-win-0.1.217-1/virtio-win-0.1.217.iso
Where are you in the installing process?

FlameKat53, 2022-04-22:
If you're on the desktop now, you can use these instructions:

once booted, when it says press any key to continue: press a key and WAIT
now that windows is installed, you should edit the file to remove the 2 lines with the install.iso.

Now that, that's done: you can install some drivers. 

Open command prompt as administrator:

bcdedit /set TESTSIGNING ON
shutdown /r /t 0

(The second command reboots, you can just use the buttons too)

now you're rebooted into test mode: go to the file manager (explorer.exe) and find your drivers cd. 
NETKVM/w10/arm64/ (OR w11 for windows 11)
find the file that you'd usually use to install a driver, right click: install, now you have networking.

bcdedit /set TESTSIGNING OFF
shutdown /r /t 0

now test signing is off and you're back in windows 10, have fun!

Maki2007, 2022-08-14:
I have managed to get it working:)
Thanks for all the help and tips. 
1. Make sure that you are running 64 bit
2. (Optional) overclock for better performance and faster installation process
3. Install packages: sudo apt install qemu-system libvirt0 cabextract wimtools chntpw genisoimage aria2
4. Get an ISO (get an bash script from uupdump.net to get an ISO and run it)
5. Get the “firmware” from /usr/share/AAVMF and paste AAVMF_CODE.fd and AAVMF_VARS.fd into a new folder (The folder where your VM will be saved)
6. In that folder, create a vhdx virtual disk: qemu-img create -f vhdx -o subformat=fixed system.vhdx 64G (64G is the size of the disk, feel free to change it)
7. Get the drivers for Win10: https://fedorapeople.org/groups/virt/virtio-win/direct-downloads/stable-virtio/virtio-win.iso
8. Create the command to start the VM:
`#!/bin/bash
qemu-system-aarch64 \
-cpu host \
-M virt,accel=kvm \
-smp 3 \
-m 2048 \
-drive file=AAVMF_CODE.fd,format=raw,if=pflash,index=0,readonly=on \
-drive file=AAVMF_VARS.fd,format=raw,if=pflash,index=1 \
-device ramfb \
-device nec-usb-xhci \
-device usb-kbd \
-device usb-mouse \
-device usb-tablet \
-device nvme,drive=systemDisk,serial=systemDisk \
-drive if=none,id=systemDisk,format=raw,file=system.vhdx \
-device usb-storage,drive=drivers,serial=drivers \
-drive if=none,id=drivers,format=raw,media=cdrom,file=drivers.iso \
-device usb-storage,drive=install,serial=install \
-drive if=none,id=install,format=raw,media=cdrom,file=win10-arm.iso \`
9. Replace .iso in the command with the actual file names
10. -smp is cores allocated to the VM (don’t use them all or you will get an BSOD)
11. -m is allocated RAM (again don’t use it all)
12. I hope it works, I have tested the instructions on a fresh installation of “The official Raspberry Pi OS 64 BIT”
Read the instructions on my website: https://sites.google.com/view/itvi/RPI/QEMU-Win10

G4vr0ch3, 2022-09-29:
>     Read the instructions on my website: https://sites.google.com/view/itvi/RPI/QEMU-Win10

So You got it to work for Windows 10 but did you get it to work for windows 11 ?

Maki2007, 2022-09-29:
Does my instructions work?
I got it working on Windows 10 on my pi. But Windows 11 doesn’t work. I don’t know why. Some time ago I tried, so don’t remember the reason, why Windows 11 don’t work.

G4vr0ch3, 2022-09-29:
> Does my instructions work?

They do for windows 10 ! 🥳 

> I got it working on Windows 10 on my pi. But Windows 11 doesn’t work. I don’t know why. Some time ago I tried, so don’t remember the reason, why Windows 11 don’t work.

I've got the same outcome, it doesn't pass the Windows11 requirements for some reason... I'll let you know if I find something relevant !

Maki2007, 2022-09-29:
Okay, thats fine.

Maki2007, 2022-09-29:
I know there are some instructions to bypass restrictions on an ordinary x86 computer on the web. I didn’t have luck. But you could try.

Maki2007, 2022-09-29:
But Windows 11 would be useless. Installing W11 native on the my pi using WOR-flasher is useless. It is so slow. So I don’t think it would be a pleasant experience. W10 is useable on native pi. But slow in a VM. Then W11 must be slower. But good luck. If I get some time, I can try getting it running on VM

Maki2007, 2022-10-11:
But thanks for the help. I will keep you updated :-)

Sendt fra mit Apple Watch

Maki2007, 2022-10-11:
Thanks. I will keep you updated :-)

Sendt fra mit Apple Watch

Maki2007, 2022-10-11:
But thanks

Sendt fra mit Apple Watch