Botspot/wor-flasher #74: cant download OS.

Visseb, 2022-05-06:
im getting this error, running Wor installer App.

An error has occurred:
Failed to determine a hashsum for WoR PE-based installer.
URL: http://worproject.ml/dldserv/worpe/gethashlatest.php
Exiting now.

Botspot, 2022-05-06:
Update: I went to the WoR discord server and the maintainer for worproject.ml shared these two screenshots:
[image]
[image]
It would appear that someone under the impression that WoR is illegal reported the website and had it taken down. This is eerily similar to the takedown of youtube-dl last year.

If I wasn't busy right now, I would try to edit WoR-Flasher to avoid depending on worproject.ml for anything.

Visseb, 2022-05-06:
Okey, take your time, im glad if you can help 😄

Botspot, 2022-05-06:
The website maintainer has moved the website to a new domain: https://worproject.com
I've updated the scripts to reflect this. Try WoR-Flasher again. It should work now.

closed 2022-05-06 by Botspot