Botspot/wor-flasher #76: Package 'exfat-utils' has no installation candidate

Gozzeh, 2022-05-20:
I'm running ubuntu 22.04 anmd it looks like the exfat-utils package is no longer available.

Is there an easy work around for this?

Itai-Nelken, 2022-05-20:
In Ubuntu 22.04 `exfat-utils` is replaced with `exfatprogs`.
See https://askubuntu.com/questions/1403900/how-to-install-exfat-utils-and-hddtemp-on-ubuntu-22-04

Gozzeh, 2022-05-20:
but what do i need to change to get it to use that and stop using
exfat-utils ...?

Itai-Nelken, 2022-05-22:
Replace exfat-utils with exfatprogs in line 239 of install-wor.sh: 
https://github.com/Botspot/wor-flasher/blob/main/install-wor.sh#L239

That should fix it.

Gozzeh, 2022-05-22:
Thanks!

I ended up just removing the exfat-utils and i got it to work but your
method seems far more sensible!

Itai-Nelken, 2022-05-22:
@Botspot a check should be added to install exfatprogs on ubuntu 22.04.

Botspot, 2022-05-22:
> @Botspot a check should be added to install exfatprogs on ubuntu 22.04.

I was under the impression that exfatprogs had to be used differently and it would involve additional changes to the script. Since I don't use Ubuntu, I had no way to know if it worked.
But if exfatprogs is a drop-in replacement for exfat-utils, that would be great!

Botspot, 2022-05-23:
I've pushed a proper fix to the script.

commit e934b6ec (2022-05-23): Prefer installing exfatprogs if possible - to close #76
closed 2022-05-23 by Botspot