Botspot/wor-flasher #77: Failed to make GPT partition table

Cpmjaguar1234, 2022-06-04:
I already made a windows 11 image for rpi4 so I reused the same one and it asked for my root password and gave me an gpt error. It said Error: Partition(s) on /dev/sdc are being used. Failed to make GPT partition table on /dev/sdc Please help.

Botspot, 2022-06-05:
It would seem that /dev/sdc somehow still had a mounted partition.

Before writing a partition table, WoR-Flasher first attempts to unmount al;l partitions in the drive to avoid precisely this error. Somehow that unmounting process failed.
If you still have your destination drive plugged in, could you try running these commands to identify the issue?
```bash
get_partition() { #Input: device & partition number. Output: partition /dev entry
  [ -z "$1" ] && error "get_partition(): no /dev device specified as"' $1'
  [ -z "$2" ] && error "get_partition(): no partition number specified as"' $2'
  [ ! -b "$1" ] && error "get_partition(): $1 is not a valid block device!"
  
  if [ "$2" == 'all' ];then
    #special mode: return every partition if $2 is 'all'
    lsblk -no path "$1" | grep -vx "$1"
  else #provided with partition number
    #list drive and partitions in $1, filter out the drive, then get the Nth line
    lsblk -no PATH "$1" | grep -vx "$1" | sed -n "$2"p
  fi
}
get_partition "$DEVICE" all
sudo umount -ql $(get_partition "$DEVICE" all)
```

Cpmjaguar1234, 2022-06-05:
I have no idea how to run the commands. I am so sorry for making you go through the trouble of helping me. I am currently on zorin os.

Cpmjaguar1234, 2022-06-07:
nvm

closed 2022-06-07 by Cpmjaguar1234

Cpmjaguar1234, 2022-06-08:
I changed my usb drive and restarted the process and it worked. Thanks botspot.