Botspot/wor-flasher #79: windows image wont boot

Zbrooklyn, 2022-06-13:
## Setup:

- Which version of Windows you choose: windows 11
- Raspberry Pi model: CM4 (same chipset as the pi 4)
- Name of SD card you tried to flash: PNY elite (I've been using these cards to boot all my OS)
- What day you first downloaded wor-flasher: 06/12/2022
- What Linux operating system you are using to run wor-flasher on: Raspberry Pi OS

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

I followed all the prompts and went through the installation process, I added "dtoverlay=dwc2,dr_mode=host" to the txt config on top of where it said do not edit below here, then continued to where it started downloading and setting up the boot image. waiting for the prompt that said to boot the pi with the new sd card. I turned off the pi and put the new sd card in. it showed the screen that said "enter to boot" the screen did not respond after 10 min it started booting then I got a blue windows error screen saying there was an error and displayed a QR code. it got stuck in the cycle trying to boot and then showing the error.

after that, I tried putting the config text I added it on the bottom of everything else and it still didn't work

Botspot, 2022-06-13:
Your setup is rather unique. I'm not sure if the config.txt changes you made caused your problem, or if it's due to the hardware you're using. Some brands of SD cards seem to trigger blue-screens for no apparent reason too.

When it comes to closed-source software like Windows, everything is awesome until it's not. And when something breaks, you can't really do much unless you work for Microsoft. :/

closed 2022-06-13 by Botspot

Zbrooklyn, 2022-06-13:
I guess I need to try without any configuration changes and I also need to try on raspberry pi 4

Zbrooklyn, 2022-06-16:
@Botspot update I did a fresh flash no modifications on raspberry pi 4 and the sd card worked with windows 11. But it does not work on cm4 even though it's the same chipset.

It's supposed to work the same ar raspberry pi 4.

Botspot, 2022-06-16:
@Zbrooklyn, I went ahead and asked a maintainer of WoR if the CM4 was compatible.
He had this to say:
[image]

Zbrooklyn, 2022-06-17:
@Botspot thank you so much