Botspot/wor-flasher #80: Wifi not showing

Zbrooklyn, 2022-06-16:
I'm using a raspberry pi 4. I installed windows using the flasher and did not modify configuration file.

The only network option it ethernet even though raspberrypi 4 has wifi and Bluetooth. Do I need a special driver?

Botspot, 2022-06-16:
WiFi driver is unavailable. Please read [the Useful information heading](https://github.com/Botspot/wor-flasher#useful-information) in the readme.

closed 2022-06-16 by Botspot

Zbrooklyn, 2022-06-21:
Will a wifi dongle work?

Botspot, 2022-06-21:
> Will a wifi dongle work if it comes with drivers? @Botspot

Previously, the answer to this question was "no", but recently someone found a USB Wi-Fi dongle that does have a compatible driver. It had to be compiled, but is now available for anyone to try out.
You would need a specific dongle for this driver to be any good.

Source: https://worproject.com/faq#usb-wi-fi-drivers