Botspot/wor-flasher #83: exfat-utils error on wor install gui

FriendlyNGeeks, 2022-06-24:
## Setup:

- Which version of Windows you chose
- Raspberry Pi model
- Name of SD card you tried to flash (some are too slow or too small)
- What day you first downloaded wor-flasher
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

missing lines for package logic like in the non graphic install

#install exfat partition manipulation utility. exfatprogs replaces exfat-utils, but they cannot both be installed at once.
if package_available exfatprogs && ! package_installed exfat-utils ;then
  install_packages exfatprogs || exit 1
else
  install_packages exfat-utils || exit 1
fi

Botspot, 2022-07-01:
> missing lines for package logic like in the non graphic install
> 
> ```shell
> #install exfat partition manipulation utility. exfatprogs replaces exfat-utils, but they cannot both be installed at once.
> if package_available exfatprogs && ! package_installed exfat-utils ;then
>   install_packages exfatprogs || exit 1
> else
>   install_packages exfat-utils || exit 1
> fi
> ```

What do you mean? Could you elaborate on this? Was there an error, or do you spot an issue in the logic here?

FriendlyNGeeks, 2022-07-01:
Error on latest ubuntu because you added lines to compensate for exfatprogs
in the non graphic version but your gui sh doesnt check for it so it errors
if the replacement package is installed and says packaged cant be installed
but is referenced.

SSG Lester, Anthony (Honorably Discharged)

friendlyneighborhoodgeeks.com

Itai-Nelken, 2022-07-01:
Thr problem is that the gui always installs exfat-utils, only the non-gui version was fixed.
https://github.com/Botspot/wor-flasher/blob/main/install-wor-gui.sh#L50