Botspot/wor-flasher #85: Will not boot with no green light

jts0, 2022-06-24:
## Setup:

- Which version of Windows you chose
 11 (10 wouldnt download)
- Raspberry Pi model
 3b v1.2
- Name of SD card you tried to flash (some are too slow or too small)
Sandisk MicroSDXC 256GB Nintendo Switch Edition
- What day you first downloaded wor-flasher
Re downloaded today
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
Raspberry Pi OS 32 bit
## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
Follow instructions just wont boot no green light

FlarosOverfield, 2022-07-16:
Same issue. I don't know why I even bother with ARM based devices. With a 50$ x86 PC I could have done this in minutes.

Botspot, 2022-07-16:
Pi 3 has been having issues lately. Even if it worked, it would be fairly useless already due to the 1 GB of RAM.

Pi 4 is recommended.

FlarosOverfield, 2022-07-16:
Yeah, because testing x86 emulation and building an SMB is sooooo useless. Also windows 10 runs on 512MB of RAM if you use the Super Charger i made.

charles-gray, 2022-12-29:
+1 on this issue. Whatever the `peinstaller/pi3/gptpatch.img` trick is to try and make GPT look like MBR doesn't seem to work for rpi 3. I tried a few variants of this manually and only succeeded in creating corrupt partition tables that nothing could read.

The workaround I'm using for anyone else... get a separate SD card, and format it as MBR/FAT-32 and just copy the EFI files to it from here: https://github.com/pftf/RPi3/releases

If you then have install-wor run on another SD card in a USB card reader, or just a USB device, plugged into the rpi USB ports, EFI on the built-in SD card slot will find the installer on the 2nd device and run from there.

I still can't get my rpi 3 to install, but at least I can spend hours in the installer trying over and over again 😆

Botspot, 2024-02-10:
Pi3 should work, now that I made it download the right uefi files for pi3. For a while it was using pi4 uefi files preventing the pi3 from booting.

closed 2024-02-10 by Botspot