Botspot/wor-flasher #87: Unable to install on Ubintu 22.04 due to deprecated exfat-utils package

grinux, 2022-06-29:
Setup sh stops with error.
Details is here:
https://askubuntu.com/questions/1403900/how-to-install-exfat-utils-and-hddtemp-on-ubuntu-22-04

Botspot, 2022-07-01:
> Setup sh stops with error.

What is the actual error? I am aware that exfat-utils is being replaced by exfatprogs, but the `install-wor.sh` script should work around that.
Here is the code used:
```bash
#install exfat partition manipulation utility. exfatprogs replaces exfat-utils, but they cannot both be installed at once.
if package_available exfatprogs && ! package_installed exfat-utils ;then
  install_packages exfatprogs || exit 1
else
  install_packages exfat-utils || exit 1
fi
```

Botspot, 2022-07-01:
See: https://github.com/Botspot/wor-flasher/issues/76

Botspot, 2022-07-01:
This has been fixed in commit https://github.com/Botspot/wor-flasher/commit/b9e099331cdabee003f22da3a9b59ad7924a8e22

closed 2022-07-01 by Botspot